e2nIEE/pandapower #1592: PandaPower Toolbox.py get_connected_elements for respect_in_service = True leads to error

AnkurArohi, 2022-05-20:
andapower has a bug in the toolbox.py function where in for the respect_in_service = True option as parameter to find all the connected elements to a bus leads to the following error.
This is also the case for simple pandapower networks
Steps to reproduce:-
Load any exemplary pandapower network
Set at the bus of intrest a generator out of service
Call the folloeing function for that bus

elems_dict = pp.get_connected_elements_dict(network , bus, respect_switches=False,respect_in_service=True)

Error
packages/pandapower/toolbox.py", line 2894, in get_connected_elements
connected_elements -= set(element_table[~element_table.in_service].index)
UnboundLocalError: local variable 'element_table' referenced before assignment

rbolgaryn, 2022-05-24:
Yes please!

rbolgaryn, 2022-05-24:
You can make a fork of the repository in your GitHub account. Then you can push your changes to your fork of pandapower as a new branch. Afterwards, you can open a pull request and specify your feature-branch as the source branch and "develop" as the desctination branch.

rbolgaryn, 2022-05-24:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

AnkurArohi, 2022-05-25:
@rbolgaryn 
Done

AnkurArohi, 2022-05-27:
@rbolgaryn 
Tests are working now

AnkurArohi, 2022-06-16:
ok, I can try

AnkurArohi, 2022-06-26:
Thanks @rbolgaryn

PR #1618 (2022-06-27): bugfix for get_connected_elements_dict; fixed FutureWarning; fixes #1592
fixes #1592 

Added test  that reproduces the issue

Fixed the bug in the function get_connected_elements
commit 13b5f97c (2022-06-27): bugfix for get_connected_elements_dict; fixed FutureWarning; fixes #1592
commit c07da310 (2022-06-27): Merge pull request #1618 from rbolgaryn/bugfix_toolbox_connected_elements

bugfix for get_connected_elements_dict; fixed FutureWarning; fixes #1592
commit f09bc915 (2022-06-27): bugfix for get_connected_elements_dict; fixed FutureWarning; fixes #1592
closed 2022-06-27 by SteffenMeinecke