e2nIEE/pandapower #1596: Geopandas not included in Anaconda package

zoltanmaric, 2022-05-29:
Hi gang! I was trying to follow the tutorial installing `pandapower` with Anaconda and running the tests, but it failed with the error below. After running `conda install geopandas`, it works fine. I guess `geopandas` should be added as a dependency to the Anaconda package, or installing it should be added to the tutorial.

```python
../../mambaforge/envs/zeroemission/lib/python3.9/site-packages/pandapower/test/api/test_auxiliary.py .........
../../mambaforge/envs/zeroemission/lib/python3.9/site-packages/pandapower/test/api/test_convert_format.py .........................
../../mambaforge/envs/zeroemission/lib/python3.9/site-packages/pandapower/test/api/test_create.py .....................
../../mambaforge/envs/zeroemission/lib/python3.9/site-packages/pandapower/test/api/test_diagnostic.py ......................
../../mambaforge/envs/zeroemission/lib/python3.9/site-packages/pandapower/test/api/test_file_io.py .......x.....x.......F

================================================================================= FAILURES ==================================================================================
_________________________________________________________________________ test_empty_geo_dataframe __________________________________________________________________________

    @pytest.mark.skipif(GEOPANDAS_INSTALLED, reason="requires the GeoPandas library")
    def test_empty_geo_dataframe():
        net = pp.create_empty_network()
        net.bus_geodata['geometry'] = None
>       net.bus_geodata = gpd.GeoDataFrame(net.bus_geodata)
E       NameError: name 'gpd' is not defined

../../mambaforge/envs/zeroemission/lib/python3.9/site-packages/pandapower/test/api/test_file_io.py:395: NameError
```

zoltanmaric, 2022-05-29:
Hmm actually, according to that skip annotation, I guess these tests should have been skipped if `geopandas` is not installed, but they weren't 🤔 I'll investigate

SteffenMeinecke, 2022-06-02:
According the last comment: correct.
According the raised issue topic: It is pandapower's policy to not require every external package which is used at some ends of the pandapower package. You can watch the install requirements at https://github.com/e2nIEE/pandapower/blob/develop/setup.py#L45-L65. If you like to install pandapower with all optional packages try `pip install pandapower --all`.

rbolgaryn, 2022-06-15:
conda-forge includes geopandas

closed 2022-06-15 by rbolgaryn