e2nIEE/pandapower #1599: Running Time series simulation of the European Low Voltage Test Feeder with 3 phase asymmetric loads

nikofou, 2022-06-02:
Hello,
i am new to pandapower and i wanted to run a timeseries simulation for the European Low Voltage Test Feeder Distribution Network (https://cmte.ieee.org/pes-testfeeders/resources/) (https://pandapower.readthedocs.io/en/v2.9.0/networks/3phase_grids.html).
As i was reading through the documentation and the tutorials, i noticed that when running the time series simulation a datasource file must be provided describing the changed in load or generation for each timestep. 
But as this network is a three phase asymmetric network what form does the datasource dataframe take? Is there a predefined template?
Also for this particular network i tried using the data contained in the .zip file in the ieee site for the load profiles. 
Thank you in advance.

elodin, 2022-06-15:
Hi @nikofou,
welcome to pandapower.

To keep things uncomplicated, i'd recommend you use one datasource for each phase. Your DataFrames will look exactly the same as the ones in the tutorial - you only use three of them instead of one.
That could look like this:

```
import numpy as np
import pandas as pd

import pandapower.control as control
import pandapower.networks as nw
import pandapower.timeseries as timeseries
from pandapower.timeseries.data_sources.frame_data import DFData

net = nw.ieee_european_lv_asymmetric(scenario='on_peak_566')

n_ts = 12

df_a = pd.DataFrame(np.random.normal(1., 0.1, size=(n_ts, len(net.asymmetric_load.index))),
                    index=list(range(n_ts)), columns=net.asymmetric_load.index) * net.asymmetric_load.p_a_mw.values

df_b = pd.DataFrame(np.random.normal(1., 0.1, size=(n_ts, len(net.asymmetric_load.index))),
                    index=list(range(n_ts)), columns=net.asymmetric_load.index) * net.asymmetric_load.p_b_mw.values

df_c = pd.DataFrame(np.random.normal(1., 0.1, size=(n_ts, len(net.asymmetric_load.index))),
                    index=list(range(n_ts)), columns=net.asymmetric_load.index) * net.asymmetric_load.p_c_mw.values

ds_a = DFData(df_a)
ds_b = DFData(df_b)
ds_c = DFData(df_c)

const_load_a = control.ConstControl(net, element='asymmetric_load', element_index=net.asymmetric_load.index,
                                    variable='p_a_mw', data_source=ds_a, profile_name=net.asymmetric_load.index)

const_load_b = control.ConstControl(net, element='asymmetric_load', element_index=net.asymmetric_load.index,
                                    variable='p_b_mw', data_source=ds_b, profile_name=net.asymmetric_load.index)

const_load_c = control.ConstControl(net, element='asymmetric_load', element_index=net.asymmetric_load.index,
                                    variable='p_c_mw', data_source=ds_c, profile_name=net.asymmetric_load.index)

ow = timeseries.OutputWriter(net, output_path="./", output_file_type=".xlsx")

ow.log_variable('res_line', 'loading_percent')

timeseries.run_timeseries(net)
```

zuno1999, 2024-09-04:
Hi @nikofou, I am also trying to add 3-ph loads to the model. Do you have any suggestions?