e2nIEE/pandapower #1612: How to modify the voltage per unit (vm_pu) of an existing grid

EIHABALMUKASHFI, 2022-06-10:
Hello everyone,
I'm trying to modify the max and min vm_pu of IEEE EUOREPAN LV NETWORK (ieee_european_lv) to be 1.1 and 0.94 respectively. I used this code 

max_vm_pu = 1.1
min_vm_pu = 0.94

net =nw.ieee_european_lv_asymmetric("off_peak_1440")
net.bus.loc[[i for i in range(len(net.bus.index))],"max_vm_pu"]=[max_vm_pu for i in range(len(net.bus.index))]
net.bus.loc[[i for i in range(len(net.bus.index))],"min_vm_pu"]=[min_vm_pu for i in range(len(net.bus.index))]
and the limits of vm_pu is  is adjusted to be equals to the inputs. However, it seems that pandapower dosent consider the updated vm_pu values since the voltage constraint is violated and the code works without errors ( continue_on_divergence=False). Can you please help me deal with this problem? thanks in advanced

rbolgaryn, 2022-06-15:
Hi @EIHABALMUKASHFI ,

take a look at the tutorial for OPF:

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/opf_basic.ipynb

Roman

closed 2022-06-15 by rbolgaryn

EIHABALMUKASHFI, 2022-06-15:
net =nw.ieee_european_lv_asymmetric("off_peak_1440")
# net.bus.loc[[i for i in range(len(net.bus.index))],"max_vm_pu"]=[max_vm_pu for i in range(len(net.bus.index))]
# net.bus.loc[[i for i in range(len(net.bus.index))],"min_vm_pu"]=[min_vm_pu for i in range(len(net.bus.index))]

net.bus["min_vm_pu"] = 1.0
net.bus["max_vm_pu"] = 1.02

index =index=np.array([755, 614, 860, 327, 813, 906, 337, 522, 813, 502, 178, 817, 896,
        83, 676, 225, 249, 701, 314, 780, 682, 639, 556,  34, 860, 342,
       320, 208, 780, 314,  70, 264, 502, 337, 614, 619, 248, 342, 289,
        34, 406, 861, 276, 900, 861, 563, 539, 817, 289, 406, 249, 320,
       886, 264, 688, 611, 562, 898, 458, 387, 388, 327, 682, 349, 778,
       899,  70, 886,  74,  73, 539, 562, 702, 702, 906, 676, 896, 349,
       611, 178, 248, 785, 388,  73, 639, 387, 785, 629,  74, 208, 522,
       556, 835, 619, 225, 688, 898,  47, 899, 563])

for i in range(55):
    net.asymmetric_load.drop(index=[i],inplace=True)

for i in range(N):
    pp.create_load(net,bus=index[i],p_mw=1)

ds_load = DFData(df_load)

ConstControl(net, "load", "p_mw", element_index=net.load.index, profile_name=col_name,data_source=ds_load)    

ow = OutputWriter(net)
run_timeseries(net,time_steps=range(1440))

I attached two images , for the output of net.bus and net.res_bus respectively . Please notice that the constraint ( max_vm_pu = 1.02) is violated  
[image]
[image]