e2nIEE/pandapower #1622: HOW TO CALCULATE THE SENSITIVITY MATRIX OF THE NETWORK

EIHABALMUKASHFI, 2022-06-25:
Is it possible to calculate the sensitivity matrix in pandapower? , I'm trying to find out how the node's voltage is sensitive to the variation of the active power in the other nodes. Then, I will apply active power curtailment to the prosumers who cause the voltage rise accordingly. To be more specific, I'm using IEEE European LVDN. I've 55 nodes with connected prosumers. Therefore, in order to avoid the voltage constraint violation, active power curtailment should be applied. In this context, how to calculate the sensitivity matrix. Thanks in advance

AnkurArohi, 2022-06-26:
Hi,
What pandapower has is makePTDF matrix which finds sensitivities of active power from nodes/elements to other branches/elements.
But this would not give you the voltage sensitivities with respect to active power changes.
To be able to correctly do this you need jacobian matrix sensitivities dv/dp.

What you can do is to extract those values from jacobian matrix directly.
For this you have to execute Newton Raphson and extract the diagonal elements after inversion.

Please do comment if you were able to analyse and do this.

My suggestion is to take a small network and debug Newton raphson matrix creation, then extract those jacobian matrix ratios.

Pls post the analysis I am intrested in this too

ascheidl, 2022-06-26:
maybe this helps:

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

AnkurArohi, 2022-06-26:
Exactly 
You need to be careful with the bus lookup abd the jacobian matrix , try to see how is the indexing(mapping) for each bus done.

EIHABALMUKASHFI, 2022-06-26:
@AnkurArohi , @ascheidl   Thanks a lot, your feedbacks would be helpful. I have another question, since I'm working with timeseries , how to get the jacobian matrix for every time slot?

AnkurArohi, 2022-06-26:
>  @AnkurArohi , @ascheidl   Thanks a lot, your feedbacks would be helpful. I have another question, since I'm working with timeseries , how to get the jacobian matrix for every time slot?

That's a good question,
It depends on your software architecture.
As pandapower has jacobian matrix as an internal function output, if there is no way of indexing and extracting the required entities, you will have to develop your own lookup, try to find out where jacobian matrix is built and port your own method out of it with the extraction possibility

AnkurArohi, 2022-06-26:
Alternative you can request a debug analysis method for pandapower so that in debug mode pandapower returns this matrix

rbolgaryn, 2022-06-27:
you can either define your own "run" function that replaces pp.runpp and pass it as the "run" parameter to run_timeseries, or define your own controller for that (see the tutorial "defining a controller").