e2nIEE/pandapower #1626: Not a bug : Can pp handle multiple control on trafo end

AnkurArohi, 2022-06-30:
Lets say we have a trafo and on one end of this trafo there is a ratio tap changer control as well as phase shift control

Can this be handled ?

rbolgaryn, 2022-06-30:
take a look at the parameter tap_shift_degree

AnkurArohi, 2022-07-01:
@rbolgaryn Thanks for your reply, yes this i am aware of but for instance the other important inputs for calculation similar to the ratio tap changer cannot be set as they are already set by the Ratio tap changer

These are 
tap_max                  
tap_min                   
tap_neutral              
tap_pos                  
shift_degree        
tap_step_percent     

shift_degree I can take over as it does not overlap with Ratio tap changer properties.
But the other are also relevant for Ratio tap changer and this is where we have an issue

AnkurArohi, 2022-07-01:
For example the trafo modelling will pick up only one tap position but if we have two controllers then (Ratio and Phase shift) then it has to be picked up from both, this is the problem

AnkurArohi, 2022-07-03:
@rbolgaryn  I dived in more, considering the documentation https://pandapower.readthedocs.io/en/v2.6.0/elements/trafo3w.html
I found two things

**Trafo3w**
There is no explanation of the usage for Phase shift values in case of trafo3w, from the formula I could understand that _only ratio tap changer/longitudnal (Längs)_ is being supported, 
So, Is Quer regelung/Logitudnal Control and/or Phase shift control not supported for trafo3w?

**Trafo2W**
As for Transverse (Quer) control in case of trafo2w documentation we need to use this parameter

**tap_step_degree**

But in the above case the solution is _limited to the same tap_neutral and tap_pos_
So, for example if the trafo has different tap positions and neutral positions for ratio tap changer and Phase shifter(Active control), the additional angle change calculation will not work, it can only work if both controls have the same tap position and tap_neutral.

Similarly in case of Ideal Phase shift trafo, as mentioned in the documentation , one can have either Phase shifter or ratio tap changer but not both and the distinguishing factor is parameter
 **tap_phase_shifter** 
Then the angle change parameter to be defined are
**tap_st_degree** , or,
**tap_step_percent** this is the same also for Longitudnal control(Längs Regelung/Ratio Tap changer)

rbolgaryn, 2022-07-05:
With the existing implementation, you could model an ideal phase shifter in series to a transformer with a longitudinal tap shifter.

rbolgaryn, 2022-07-05:
Otherwise, there needs to be an adjustment to the implementation.

AnkurArohi, 2022-07-05:
> Dear @AnkurArohi ,
> 
> for trafo3w, the phase shift is not implemented.
> 
> For trafo 2W - is it common for transformers to have unequal tap positions separately for longitudinal and transverse control?
> 
> Roman

Hi @rbolgaryn 
Thank you for the reply. 
Yes sure, currently I have these parameters for one trafo end
[image]
but this one if for trafo 2W

rbolgaryn, 2022-07-05:
Thanks for the info. 
Currently, the trafo3w is not implemented for phase shifting.

Do you know if it is also common for trafo 2W to have different tap parameters for longitudinal and transverse?

AnkurArohi, 2022-07-05:
<cim:RatioTapChanger rdf:ID="_11001aaa-a866-5849-8346-9d3379a3e8f2">
		<cim:IdentifiedObject.name>TRA 101 (PST)</cim:IdentifiedObject.name>
		<cim:TapChanger.highStep>35</cim:TapChanger.highStep>
		<cim:TapChanger.lowStep>1</cim:TapChanger.lowStep>
		<cim:TapChanger.ltcFlag>false</cim:TapChanger.ltcFlag>
		<cim:TapChanger.neutralStep>18</cim:TapChanger.neutralStep>
		<cim:TapChanger.neutralU>117.0000000000</cim:TapChanger.neutralU>
		<cim:TapChanger.normalStep>18</cim:TapChanger.normalStep>
		<cim:RatioTapChanger.tculControlMode rdf:resource="http://iec.ch/TC57/2013/CIM-schema-cim16#TransformerControlMode.volt"/>
		<cim:RatioTapChanger.stepVoltageIncrement>-0.9411764741</cim:RatioTapChanger.stepVoltageIncrement>
		<cim:RatioTapChanger.TransformerEnd rdf:resource="#_6b59df68-630f-5013-892a-17ee91dfd6da"/>
	</cim:RatioTapChanger>

	<cim:PhaseTapChangerTabular rdf:ID="_691b6bb5-05e3-515a-854c-db2e2f873e0e">
		<cim:IdentifiedObject.name>TRA 101 (PST)</cim:IdentifiedObject.name>
		<cim:TapChanger.highStep>30</cim:TapChanger.highStep>
		<cim:TapChanger.lowStep>1</cim:TapChanger.lowStep>
		<cim:TapChanger.ltcFlag>true</cim:TapChanger.ltcFlag>
		<cim:TapChanger.neutralStep>18</cim:TapChanger.neutralStep>
		<cim:TapChanger.neutralU>117.0000000000</cim:TapChanger.neutralU>
		<cim:TapChanger.normalStep>18</cim:TapChanger.normalStep>
		<cim:PhaseTapChanger.TransformerEnd rdf:resource="#_ba8093ed-2c96-5c28-aa5a-e9fa2f13658d"/>
		<cim:PhaseTapChangerTabular.PhaseTapChangerTable rdf:resource="#_6b3cac0c-400e-5d19-b171-01f1315938f3"/>
	</cim:PhaseTapChangerTabular>

rbolgaryn, 2022-07-05:
I see.

We don't support such a case at the moment. The implementation for that case would have to be adjusted.

Roman

AnkurArohi, 2022-07-05:
But if the neutral step for both are same then atleast Longitudnal and Latitudnal cases can work, but PST not
For PST it has be on a different end 
@rbolgaryn  then it should work right?
Or, as I am trying now to implement ideal PST in series , but this will lead to changed power flow results (a bit)

AnkurArohi, 2022-07-06:
Dear @rbolgaryn 
Do you know what exactly is the difference between Longitudnal control(PP documentation) and Phase shift control, In CGMES they are both the same but in PP Documentation they are differently handled and the details I have already mentioned above. Can you please let me know? Thanks

AnkurArohi, 2022-07-06:
To put in other words the 

shift_degree parameter is used when pf is run in voltage angle true Mode

 then for each step there is also a value of tap_step_degree that one can define for longitudnal Regulator.

But longitudnal Regulator cannot work if the neutral pos and current pos of tap is different as compared to latitudnal  Regulator because Trafo can have only one tap_pos which is coming from ratio/latitudnal control. I think this should be mentioned in pp document.

Is this correct interpretation?

Another issue
Let's say I have the value of shift degree which changes with each tap_pos and I am using it for voltage angle based of now if for ideal pst I define tap_step_degree, what changes here?