e2nIEE/pandapower #1630: consistency in naming: element, index, element_index, element_type, ...

SteffenMeinecke, 2022-07-06:
1)
Some examples of current code:

* poly_cost/pwl_cost: element, et
* switch: element, et
* const_control: element_index, element
* measurement: element, element_type

Proposual by @rbolgaryn and @SteffenMeinecke for consistency:
* element, element_type

2)
Switch support of `net.switch.et` from "t", "t3", "l", ... to "trafo", "trafo3w", "line", ...

SteffenMeinecke, 2024-07-08:
Now, my proposual is `element_type` and `element_index` (least misleading, I think).
With https://github.com/e2nIEE/pandapower/pull/2334, group gets this proposed names while controllers have first consistent names `element` and `element_index`. Adapting the controllers and all others to `element_type` and `element_index` were to effortful to me at the moment.

vogt31337, 2025-11-26:
Would be nice to implement, but it will create a lot of problems, issues and incompatibilities. Therefore, I will close this issue for now, since these things are documented.