e2nIEE/pandapower #1638: Exception: EXCEPTION_ACCESS_VIOLATION when running timeseries opf with pandamodels

angelinasyrri, 2022-07-19:
Hi! I am running a timeseries opf with pandamodels:

pp.timeseries.run_timeseries(net_flex, time_steps = range(0, n_timesteps_peak),run = pp.runpm_ac_opf,continue_on_divergence=True, delta=1e-16)

while everything seems to run smooth, after a certain number of timesteps (around 60% of the total period I am examining), I get the following error:
(I am using a Windows machine, python installation, and Spyder standalone interface). I understand it is an issue related to Julia  ( I am using Julia 1.5.4), but I have no idea how to proceed. Any help would be more than welcome!

_Please submit a bug report with steps to reproduce this fault, and any error messages that follow (in their entirety). Thanks.
Exception: EXCEPTION_ACCESS_VIOLATION at 0x2d016c5 -- is_ir_node at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:493 [inlined]
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:570
in expression starting at none:0
is_ir_node at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:493 [inlined]
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:570
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:766 [inlined]
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:767
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:531 [inlined]
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:700
jl_serialize_value_ at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:528 [inlined]
jl_compress_ir at /cygdrive/c/buildbot/worker/package_win64/build/src\dump.c:2601
cache_result at .\compiler\typeinfer.jl:122
typeinf at .\compiler\typeinfer.jl:67
typeinf_edge at .\compiler\typeinfer.jl:484
abstract_call_method at .\compiler\abstractinterpretation.jl:419
abstract_call_gf_by_type at .\compiler\abstractinterpretation.jl:111
abstract_call_known at .\compiler\abstractinterpretation.jl:904
abstract_call at .\compiler\abstractinterpretation.jl:926
abstract_apply at .\compiler\abstractinterpretation.jl:635
abstract_call_known at .\compiler\abstractinterpretation.jl:704
abstract_call at .\compiler\abstractinterpretation.jl:926
abstract_call at .\compiler\abstractinterpretation.jl:911
abstract_eval at .\compiler\abstractinterpretation.jl:1005
typeinf_local at .\compiler\abstractinterpretation.jl:1270
typeinf_nocycle at .\compiler\abstractinterpretation.jl:1326
typeinf at .\compiler\typeinfer.jl:12
typeinf_ext at .\compiler\typeinfer.jl:570
typeinf_ext at .\compiler\typeinfer.jl:601
jfptr_typeinf_ext_24535.clone_1 at C:\Users\angel\AppData\Local\Programs\Julia 1.5.4\lib\julia\sys.dll (unknown line)
_jl_invoke at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:2214 [inlined]
jl_apply_generic at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:2398 [inlined]
jl_apply at /cygdrive/c/buildbot/worker/package_win64/build/src\julia.h:1691 [inlined]
jl_type_infer at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:296
jl_generate_fptr at /cygdrive/c/buildbot/worker/package_win64/build/src\jitlayers.cpp:290
jl_compile_method_internal at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:1964
jl_compile_method_internal at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:1931 [inlined]
_jl_invoke at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:2224 [inlined]
jl_apply_generic at /cygdrive/c/buildbot/worker/package_win64/build/src\gf.c:2398
jl_apply at /cygdrive/c/buildbot/worker/package_win64/build/src\julia.h:1691 [inlined]
do_apply at /cygdrive/c/buildbot/worker/package_win64/build/src\builtins.c:674
jl_f__apply at /cygdrive/c/buildbot/worker/package_win64/build/src\builtins.c:688 [inlined]
jl_f__apply_latest at /cygdrive/c/buildbot/worker/package_win64/build/src\builtins.c:724
#invokelatest#1 at .\essentials.jl:710 [inlined]
invokelatest at .\essentials.jl:709
jl_apply at /cygdrive/c/buildbot/worker/package_win64/build/src\julia.h:1691 [inlined]
do_apply at /cygdrive/c/buildbot/worker/package_win64/build/src\builtins.c:674
_pyjlwrap_call at C:\Users\angel\.julia\packages\PyCall\7a7w0\src\callback.jl:28
unknown function (ip: 00000180f363cc94)
pyjlwrap_call at C:\Users\angel\.julia\packages\PyCall\7a7w0\src\callback.jl:44
unknown function (ip: 0000017eb42b0b4f)
unknown function (ip: 0000017eb41f41f9)
unknown function (ip: 0000017eb41f4062)
PyEval_EvalFrameDefault at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyFunction_Vectorcall at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyThreadState_Get at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyEval_EvalFrameDefault at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyFunction_Vectorcall at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyEval_EvalFrameDefault at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyFunction_Vectorcall at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyEval_EvalFrameDefault at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyFunction_Vectorcall at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
PyVectorcall_Call at C:\Users\angel\AppData\Local\Programs\Python\Python310\python310.dll (unknown line)
[... 107 lines cut ...]
unknown function (ip: 00007ff74be3122f)
BaseThreadInitThunk at C:\windows\System32\KERNEL32.DLL (unknown line)
RtlUserThreadStart at C:\windows\SYSTEM32\ntdll.dll (unknown line)
Allocations: 8686054464 (Pool: 8681434739; Big: 4619725); GC: 2697

Windows fatal exception: access violation

Main thread:
Current thread 0x00001a80 (most recent call first):
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\julia\core.py", line 215 in __setattr__
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\opf\run_powermodels.py", line 87 in _call_pandamodels
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\opf\run_powermodels.py", line 34 in _runpm
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\runpm.py", line 134 in runpm_ac_opf
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\control\run_control.py", line 180 in _evaluate_net
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\control\run_control.py", line 218 in control_implementation
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\control\run_control.py", line 298 in run_control
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\timeseries\run_time_series.py", line 126 in run_time_step
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\timeseries\run_time_series.py", line 321 in run_loop
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\pandapower\timeseries\run_time_series.py", line 349 in run_timeseries
  File "c:\users\angel\...\development\python_scripts\performance_assessment_monte_carlo_greek_dso_net.py", line 1359 in <module>
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\spyder_kernels\py3compat.py", line 356 in compat_exec
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 469 in exec_code
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 611 in _exec_file
  File "C:\Users\angel\...\development\work_env_spyder\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 524 in runfile
  File "C:\Users\angel\AppData\Local\Temp\ipykernel_11812\249685542.py", line 1 in <cell line: 1>_

ZhengLiu1119, 2022-08-01:
hi angelinasyrri,

sorry for the late answer. I was on vacation. 

are there any results if you only run it for the 60% (or 50%) of the total period? If yes, as a workaround, you can split your time series into some smaller time windows. 
But, to be honest, I don't think it works with `run_timeseries(net, run=pp.runpm_ac_opf)`. Because for the processing of time series, pandapower (python) and PowerModels (Julia) are not the same. It might be not so well documented. Presently, `run_timeseries` could only support `run=pp.runpp` and `run=pp.runopp`.

I would say that if it makes sense for your use case, you can realize the time series optimization with pandamodels in a loop, e.g.,

`
from copy import deepcopy
for tp in range(10):
    # update the operating points of load and sgen
    net.load.p_mw = YOUR_DATA[("load", "p_mw")][tp]
    net.gen.p_mw = YOUR_DATA[("sgen", "p_mw")][tp]
    # create a container
    res = {}
    # run opf
    pp.runpm_ac_opf(net)
    res[tp] = deepcopy(net)
`  

currently, we are improving the interface between pandapower and PowerModels.
Later, users can deal with time-series optimization with pandamodels.

closed 2022-08-16 by ZhengLiu1119

angelinasyrri, 2022-08-22:
Hi @ZhengLiu1119 ! Sorry for the late response, it was my turn for summer vacation.  It works fine for shorter periods, therefore I also thought that I will split the run in a few parts to properly get the results. It had been a memory issue as it seems.

Other than that, you made me a bit skeptical with your comment about run_timeseries and Julia. I have been using it for a time, and it seems to give good results, so I thought it works well. I am using this command:

pp.timeseries.run_timeseries(net_flex, time_steps = range(0, n_timesteps_peak),run = pp.runpm_ac_opf,continue_on_divergence=True, delta=1e-16)

Thanks!