e2nIEE/pandapower #1639: New OPF algorithm : Second Order Cone Programming for radial networks

gcherot, 2022-07-19:
I use PandaPower for my research as I am developing new algorithms to control grid flexibilities.
I usually compare the results of my algorithms with those given by PandaPower's OPF AC.
The problem is that, as I understand it, it is solved using the interior point solver which does not guarantee optimality since the AC-OPF is not convex.

In radial networks (such as distribution networks), the second order cone programming formulation is accurate (and fast) under mild assumptions.
Have you considered implementing it?
Would you be interested in my contribution?

Thank you!

AnkurArohi, 2022-07-19:
Second order cone is a good algorithm, 
Can you send a pseudo code ?

rbolgaryn, 2022-07-20:
We would be interested in your contribution!

Please open a pull request for this implementation.

ascheidl, 2022-07-20:
pandapower has an interface to the PowerModels.jl package which includes besides SOC different other Approximations and Relaxations 

https://lanl-ansi.github.io/PowerModels.jl/stable/formulation-details/#Formulation-Details

gcherot, 2022-07-20:
Thank you for your answers!

@AnkurArohi I don't have a pseudo code yet.
@ascheidl I think that is exactly what I need. I will try it, I'll get back to you.
@rbolgaryn Since PowerModels does it, I'm not sure my contribution is needed.

closed 2022-07-27 by rbolgaryn