e2nIEE/pandapower #1640: pandapowerIEEE118 line number is not the same as pypower(true IEEE118case)

fengbin150215, 2022-07-20:
In [pandapower/networks/power_system_test_case_jsons/case118.json](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/networks/power_system_test_case_jsons/case118.json)
there is only 173 lines 

But in  [pypower/case118.py](https://github.com/rwl/PYPOWER/blob/master/pypower/case118.py) there are 186 lines(branch).

I check the [paper](https://ieeexplore.ieee.org/abstract/document/7904729) and some website they all say IEEE188 case have 186lines. 
I am wondering why the pandapower IEEE 118 case only have 173lines?

rbolgaryn, 2022-07-22:
Hi  @fengbin150215 ,

have you checked if some of the branches from pypower are represented by transformers in pandapower? (net.trafo)

Roman

fengbin150215, 2022-07-22:
OK！ Thanks! I forget the transformers part!

closed 2022-07-22 by fengbin150215