e2nIEE/pandapower #1643: DC load flow analysis

ramsingh1995, 2022-07-22:
I want to do DC load flow analysis in pandapower. Can pandapower be used for DC load flow. Kindly suggest me the open source tools for DC load flow analysis.

rbolgaryn, 2022-07-22:
if you mean DC load flow then use pp.rundcpp(net).

If you mean a power flow calculation in DC grids that considers line losses etc. then pandapower is not suitable.
A DC power flow with Newton-Raphson method can be easily implemented though.

Roman

ramsingh1995, 2022-07-22:
Thanks for the reply.
Can you please suggest me any open source tool for DC load flow analysis as I have to do laod flow analysis on a DC microgrid?

Ram

rbolgaryn, 2022-07-22:
Sorry can't help you. You can try looking up OS packages with a search engine, e.g. I found Scilab + Xcos, etc.

closed 2022-07-22 by rbolgaryn

AnkurArohi, 2022-07-22:
To be very frank you can write your own equations considering approximation
Let all voltage angles to be zero, reactive power Constant
X =0

Try to use these changes in pandapower formulas