e2nIEE/pandapower #1646: Consistency for 3 phase power-flow for transformer

nitbharambe, 2022-07-25:
Hi all,
I have bumped into a case where the power-flow in and out of a node in case of transformer is not close to zero. Below is a minimal example of that case. 

Configuration is as such: ` ext_grid---(bus1)---trafo---(bus2)---line---(bus3)--asym_load`

The powerflow out of the transformer and into the line is not zero. The difference is around 1kw and I was not able to figure out why it is so. Thanks again for your help!

```
import pandapower as pp

net = pp.create_empty_network()
bus1 = pp.create_bus(net, vn_kv=20, name="Bus 1")
bus2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")
bus3 = pp.create_bus(net, vn_kv=0.4, name="Bus 3")

ext_grid = pp.create_ext_grid(net, bus=bus1, vm_pu=1.0, s_sc_max_mva=10, rx_max=0.1, x0x_max=1, r0x0_max=0.1)

trafo = pp.create_transformer_from_parameters(net, hv_bus=bus1, lv_bus=bus2, sn_mva=30,
        vn_lv_kv=0.4, vn_hv_kv=20, vkr_percent=1.58, vk_percent=4,
        pfe_kw=0.8, i0_percent=0.32, vector_group='Dyn',
        vk0_percent=4, vkr0_percent=0.158, mag0_percent=50,
        mag0_rx=0.1, tap_phase_shifter=False,
        si0_hv_partial=0.9, in_service=True, shift_degree=150)
line = pp.create_line_from_parameters(net, from_bus= bus2, to_bus=bus3, length_km=10,
                r0_ohm_per_km=0.642, x0_ohm_per_km=0.083, c0_nf_per_km=210.0,
                r_ohm_per_km=0.642, x_ohm_per_km=0.083, c_nf_per_km=210.0, max_i_ka=0.142)
asym_load = pp.create_asymmetric_load(net, bus3, p_a_mw=0.1, p_b_mw= 0.09, p_c_mw=0.08,
                                       type='wye')

pp.runpp_3ph(net, tolerance_mva=1e-8, numba=False)

print("Consistency")
print("Trafo-line a: %.8f" % (net.res_trafo_3ph["p_a_lv_mw"][trafo] + net.res_line_3ph["p_a_from_mw"][line]))
print("Trafo-line b: %.8f" % (net.res_trafo_3ph["p_b_lv_mw"][trafo] + net.res_line_3ph["p_b_from_mw"][line]))
print("Trafo-line c: %.8f" % (net.res_trafo_3ph["p_c_lv_mw"][trafo] + net.res_line_3ph["p_c_from_mw"][line]))

```

Output:

[image]Hi all,
I have bumped into a case where the power-flow in and out of a node in case of transformer is not close to zero. Below is a minimal example of that case. 

Configuration is as such: ` ext_grid---(bus1)---trafo---(bus2)---line---(bus3)--asym_load`

The powerflow out of the transformer and into the line is not zero. The difference is around 1kw and I was not able to figure out why it is so. Thanks again for your help!

```
import pandapower as pp

net = pp.create_empty_network()
bus1 = pp.create_bus(net, vn_kv=20, name="Bus 1")
bus2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")
bus3 = pp.create_bus(net, vn_kv=0.4, name="Bus 3")

ext_grid = pp.create_ext_grid(net, bus=bus1, vm_pu=1.0, s_sc_max_mva=10, rx_max=0.1, x0x_max=1, r0x0_max=0.1)

trafo = pp.create_transformer_from_parameters(net, hv_bus=bus1, lv_bus=bus2, sn_mva=30,
        vn_lv_kv=0.4, vn_hv_kv=20, vkr_percent=1.58, vk_percent=4,
        pfe_kw=0.8, i0_percent=0.32, vector_group='Dyn',
        vk0_percent=4, vkr0_percent=0.158, mag0_percent=50,
        mag0_rx=0.1, tap_phase_shifter=False,
        si0_hv_partial=0.9, in_service=True, shift_degree=150)
line = pp.create_line_from_parameters(net, from_bus= bus2, to_bus=bus3, length_km=10,
                r0_ohm_per_km=0.642, x0_ohm_per_km=0.083, c0_nf_per_km=210.0,
                r_ohm_per_km=0.642, x_ohm_per_km=0.083, c_nf_per_km=210.0, max_i_ka=0.142)
asym_load = pp.create_asymmetric_load(net, bus3, p_a_mw=0.1, p_b_mw= 0.09, p_c_mw=0.08,
                                       type='wye')

pp.runpp_3ph(net, tolerance_mva=1e-8, numba=False)

print("Consistency")
print("Trafo-line a: %.8f" % (net.res_trafo_3ph["p_a_lv_mw"][trafo] + net.res_line_3ph["p_a_from_mw"][line]))
print("Trafo-line b: %.8f" % (net.res_trafo_3ph["p_b_lv_mw"][trafo] + net.res_line_3ph["p_b_from_mw"][line]))
print("Trafo-line c: %.8f" % (net.res_trafo_3ph["p_c_lv_mw"][trafo] + net.res_line_3ph["p_c_from_mw"][line]))

```

Output:

[image]

AnkurArohi, 2022-07-25:
are you sure with the sgift degree attribute in trafo?

nitbharambe, 2022-08-08:
> are you sure with the shift degree attribute in trafo?

Hi. I apologize, I'm not quite sure. Can the shift degree make consistency invalid?

vogt31337, 2025-04-02:
@nitbharambe yes the shift_degree value can significantly change the results.
I know it is quite old topic, but could you by any Chance Check the results with the newest PP Version?

nitbharambe, 2025-04-08:
@vogt31337 

At 2.14.10, this code gives:

Consistency
Trafo-line a: -0.00106047
Trafo-line b: 0.00036535
Trafo-line c: 0.00069513

And at 3.0, I get LoadflowNotConverged.

The grid is this way:
`ext_grid -- bus -- transformer -- bus -- line -- bus --asymmetric_load`
And I imagined the individual phases to be connected this way: 
```
transformer_out_phase_a --- line_phase_in_a
transformer_out_phase_b --- line_phase_in_b
transformer_out_phase_c --- line_phase_in_c
```
I do see that these differences found in result sum up to 0. Is this modelling I assumed incorrect?

vogt31337, 2025-04-08:
@nitbharambe have you changed the Shift degree value? It is by Standard at 150, try setting it to 0.

nitbharambe, 2025-04-08:
shift_degree=0 and everything else the same in the above code leads to non convergence in 3.0. 
And for 2.14.10 it gives:
```
Consistency
Trafo-line a: 0.06821216
Trafo-line b: -0.00635691
Trafo-line c: -0.06185796
```
I dont think I completely grasp this. There shouldn't be anything like Dyn0 transformer right? :D

Hence, I also tried with changing vector_group="YNyn" and shift_degree=0, and the code gave non convergence on both 3.0 and 2.14.10. Hence stuck over here too.