e2nIEE/pandapower #1647: Enquiry on what the 'vn_kv' parameter stands for ?

Horreyneeyo, 2022-07-26:
Hello all,

I am a new user of Pandapower. I would like to seek clarification on what 'vn_kv' means. Is it the nominal voltage or rated voltage. For example in Pandapower.networks.testcases e.g IEEE 14 bus test case, the buses have a norminal voltages of either 69kV, 13.8kV and 18kV as applicable. However, the voltage specified for the buses on pandapower are different and probably adapted with values of 135kV, 0.208 kV and 12kV.
I am confused as regards how this was derived. I would be grateful if explanation can be given or references materials are pointing out to me.

AnkurArohi, 2022-07-26:
They are rated values
(https://pandapower.readthedocs.io/en/v2.9.0/elements/bus.html)

Is there a bug difference in nominal and rated values for buses in the data model that you have?

Horreyneeyo, 2022-07-26:
Hello Ankur
[image]
[image]

Thanks for the response. 
When I crosschecked the  IEEE test cases

AnkurArohi, 2022-07-26:
I think you are comparing different files

Horreyneeyo, 2022-07-26:
Well, IEEE 14 is sub-transmission/distribution. I am comparing the network parameters of the original IEEE 14 test case applicable in Matpower and the pp.networks.case14 parameters. I understand there would be differences e.g. Matpower IEEE 14 uses per-unit parameter values while pp uses per km in the case of RXB.
I am wondering how to determine the rated voltage values as used in pp.

SteffenMeinecke, 2022-07-28:
As described in the [docs](https://pandapower.readthedocs.io/en/develop/networks/power_system_test_cases.html#case-14), in case14 and case57 assumptions must be made since the data origin have no voltage information.
@Horreyneeyo Which is the origin of the `vn_kv` data you showed?
After 5 HV node, a MV/LV node follows and then there are two nodes which are connected to the medium voltage connection of a 3winding transformer. In pp, `vn_kv` of these nodes are 14 and 12 kV. Why do you have 13.8 and 18?

Horreyneeyo, 2022-07-28:
The origin of the vn_kv (the white background pic as seen above) is from the IEEE 14 bus datasheet as seen [here](https://www.academia.edu/7781632/A_DATA_SHEETS_FOR_IEEE_14_BUS_SYSTEM).  You are correct about the 3-winding transformer connection. the values 13.8kV and 18kV are the base kV as seen in the aforementioned datasheet above.