e2nIEE/pandapower #1669: implementation of finding which elements belong to new grid equivalents should be more robust

SteffenMeinecke, 2022-08-08:
In the `if` clause of `get_equivalent()` new equivalent elements are collect to a group. This implementation is not as resistant as it could. It uses the naming convention of the internally used create functions. However, it is not checked whether this approach always works fine, e.g. if the input net has names like "..._separate". It is recommended to return the indices of the elements to allow creating a group with that, or to first create a group an then append new equivalent elements to that group.

ZhengLiu1119, 2022-08-23:
- [ Intergration to pandapower documentation is still missing]

ZhengLiu1119, 2022-08-23:
- [ ] Intergration to pandapower documentation is still missing