e2nIEE/pandapower #1673: The branch loss in Result Parameters are different from console output

Shapeno, 2022-08-10:
I tried this case in V2.10.1 , and it works well.
```python
import pandapower as pp
net = pp.create_empty_network()

#create buses
bus1 = pp.create_bus(net, vn_kv=220.)
bus2 = pp.create_bus(net, vn_kv=110.)
bus3 = pp.create_bus(net, vn_kv=110.)
bus4 = pp.create_bus(net, vn_kv=110.)

#create 220/110 kV transformer
pp.create_transformer(net, bus1, bus2, std_type="100 MVA 220/110 kV")

#create 110 kV lines
pp.create_line(net, bus2, bus3, length_km=70., std_type='149-AL1/24-ST1A 110.0')
pp.create_line(net, bus3, bus4, length_km=50., std_type='149-AL1/24-ST1A 110.0')
pp.create_line(net, bus4, bus2, length_km=40., std_type='149-AL1/24-ST1A 110.0')

#create loads
pp.create_load(net, bus2, p_mw=60., controllable = False)
pp.create_load(net, bus3, p_mw=70., controllable = False)
pp.create_load(net, bus4, p_mw=10., controllable = False)

#create generators
eg = pp.create_ext_grid(net, bus1, min_p_mw = -1e6, max_p_mw = 1e6)
g0 = pp.create_gen(net, bus3, p_mw=0, min_p_mw=0., max_p_mw=80.,vm_pu=1.01, controllable=True)
g1 = pp.create_gen(net, bus4, p_mw=0, min_p_mw=0., max_p_mw=100., vm_pu=1.01, controllable=True)

costeg = pp.create_poly_cost(net, 0, 'ext_grid', 1)
costgen1 = pp.create_poly_cost(net, 0, 'gen', 1)
costgen2 = pp.create_poly_cost(net, 1, 'gen', 1)

pp.runopp(net, verbose=True)
```
However, I found branch loss in Result Parameters are different from console output. 

-----

* **console outputs**:   
reactive power consumption of lines are 0.00, 0.00, 0.01.

```
================================================================================
|     Branch Data                                                              |
================================================================================
Brnch   From   To    From Bus Injection   To Bus Injection     Loss (I^2 * Z)  
  #     Bus    Bus    P (MW)   Q (MVAr)   P (MW)   Q (MVAr)   P (MW)   Q (MVAr)
-----  -----  -----  --------  --------  --------  --------  --------  --------
   0      1      2     -1.27     -1.18      1.28     -1.14     0.002      0.00
   1      2      3     -0.08     -0.83      0.08     -0.83     0.000      0.00
   2      3      1      2.33     -0.62     -2.33     -0.70     0.003      0.01
   3      0      1     56.53      1.98    -56.40      1.88     0.083      3.84
                                                             --------  --------
                                                    Total:     0.088      3.85
```

* **net.res_line**:   
reactive power consumption of lines are -2.320812,-1.663050,-1.321093

|   | p_from_mw | q_from_mvar | p_to_mw | q_to_mvar | pl_mw | **_ql_mvar_** |
|----|----|----|----|----|----|----|
|0|-1.274188|-1.181771|1.276017|-1.139041|0.001828|**_-2.320812_**|
|1|-0.079838|-0.831672|0.079844|-0.831378|0.000005|**_-1.663050_**|
|2|2.332971|-0.620909|-2.329480|-0.700184|0.003492|**_-1.321093_**|

Except the reactive power consumption, other outputs are the same.

SteffenMeinecke, 2022-09-28:
Hi @Shapeno , @lthurner, 
I think, the console output of ql_mvar via `verbose=True` is wrong and the pandapower result in net.res_line is correct.
The console output is based on pypower code. https://github.com/e2nIEE/pandapower/blob/develop/pandapower/pypower/printpf.py#L165-L167 results in a difference of q_from_mvar and q_to_mvar instead of a sum...

KS-HTK, 2024-11-27:
With the current development version of pandapower I get an `OPFNotConverged: Optimal Power Flow did not converge!` for the example code given here.