e2nIEE/pandapower #1677: Debugging NR non convergence case

AnkurArohi, 2022-08-17:
I have a non converging NR and I am trying to find out what could lead to this failure.
While running the diagnostic function i could find out two major points:

{'impedance_values_close_to_zero': [{'xward': [5, 31],
   'impedance': [2,
    9,
    10,
    23,
    26,
    27,
    41,
    43,
    49,
    67,
    68,
    73,
    81,
    82,
    84,
    92,
    99,
    100,
    104]},
    
    I tried to set the impedance out of service but still the network is not converging.
A further analysis showed that the same network has a NR convergence case in INTEGRAL.
Any suggestions?

rbolgaryn, 2022-08-17:
try increasing the impedance instead of setting it out of service

AnkurArohi, 2022-08-17:
i am sorry, i actually increased the impedance, i meant i tried to set xward out of service, as i see that the impedance of xward is not low,
still it does not converge

rbolgaryn, 2022-08-23:
you set the impedance of these elements higher, and still doesn't converge?
You can also try running DC power flow and seeing if any elements are overloaded to a high extent.

AnkurArohi, 2022-08-24:
yes still does not converge, DC Power flow/Initial guess converges

AnkurArohi, 2022-08-29:
Are extended wards also modelled as PV node or PQ node, as per the description in pandapower docs, I came to the conclusion that they are modelled as PQ nodes, is that right?

rbolgaryn, 2022-09-26:
Yes, ward is PQ and xward is PV

closed 2022-11-03 by AnkurArohi