e2nIEE/pandapower #1679: OPF doesn't converge when hourly load model of IEEE 14 RTS is reduced

Horreyneeyo, 2022-08-18:
Hello,

I am using the Pandapower software to perform an OPF on the IEEE 14 bus network (case14). I have been able to develop a hourly load model for a year (1 - 8760 hours) which I have converted to C language and tested on Matlab and it is working. However, I noticed that for pandapower, the 1st hour runs but the second hour gives an error (OPFNotConverged: Optimal Power Flow did not converge!) which has a generally reduced load value than the first hour. 
What can I do to ensure OPF converges? 
Does Pandapower have a version or library that caters for power system planning/ reliability assessment?

rbolgaryn, 2022-08-23:
Hello @Horreyneeyo ,

you can check if a solution is possible by manually changing the power values of generators and loads, and checking if the loadings/voltages are within limits. It is possible that the solution cannot be found. Also, if the grid has transformers, one could check if the tap positions are causing voltages to be out of range for the timestep. 
Those are just ideas, a lot is trial and error.
For reliability/planning we do not have an open source library.

Best regards,
Roman

Horreyneeyo, 2022-08-24:
Thanks for the response.
Yeah. I plan to use negative static generation(sgen) as displaceable loads on the network. However, I would like to find out if changes made in the scaling factor for sgen would also affect the max/min value of p_mw and q_mvar. I know it obviously does for the p_mw when you check it on net.res_sgen.p_mw.

Yiwen233, 2023-06-07:
I am facing the almost the same problem, runopp() did not converge whatever I change the net structure or load, but runpp() works well. So have you addressed this problem and how you do that?

Horreyneeyo, 2023-06-07:
@Yiwen233 I could not solve it. So I ended up using MATLAB/Matpower for the OPF timeseries analysis

Horreyneeyo, 2023-06-07:
To modify the Matpower OPF. Read the materials below in details and you should be able to adjust the OPF objective function as desired.
[https://matpower.org/docs/MOST-manual-1.1.pdf](url) 
[https://matpower.org/docs/TN2-OPF-Derivatives.pdf](url)
[https://matpower.org/docs/MIPS-manual-1.5.pdf](url)
[https://matpower.org/docs/[MATPOWER-manual-8.0b1.pdf](https://matpower.org/docs/MATPOWER-manual-8.0b1.pdf)](url)

Yiwen233, 2023-06-08:
> To modify the Matpower OPF. Read the materials below in details and you should be able to adjust the OPF objective function as desired. [https://matpower.org/docs/MOST-manual-1.1.pdf](url) [...]

Thanks, I'll figure it out

closed 2026-11-11 by vogt31337