e2nIEE/pandapower #1691: Are there any functions or expressions to add personal constraints for OPF?

DavidLudl, 2022-08-25:
Dear Pandapower Team,

First, thank you for your effort for this fantastic project.

While I am preparing my master thesis with pandapower, I am trying to add some further variables and constraints for the OPF problem, like adding new constraints for install capacity of wind turbine. Therefore, I would like to ask if there are any functions or expressions in pandapower to allow users to add their personal constraints? If there is no such function, where can I find and modify the constraints for OPF?

Best regards.
David

rbolgaryn, 2022-08-31:
You can takle a look at PowerModels integration and see if it is implemented.
There is a pandapower tutorial about using PowerModels as a backend

DavidLudl, 2022-10-19:
Ok, thank you for your advice, I will try to have a look of that.

SteffenMeinecke, 2023-07-12:
closed due to inactivity

closed 2023-07-12 by SteffenMeinecke