e2nIEE/pandapower #1692: How to add ramp rate constraints to the pandapower.runopp.

lalala215, 2022-08-28:
I want to slove the OPF problem with ramp rate constraints
I am wondering how to add ramp rate constraints to the pandapower.runopp? 

If anyone know this porblem, please leave a message! Thanks a lot

rbolgaryn, 2022-08-31:
I don't think you can, only indirectly by setting the max_p_mw and min_p_mw depending on the p_mw value of the previous time step. But it soes not solve the problem of optimizing across several time steps. You can check if PowerModels support this functionality and use the PowerModels impelmentation as the backend for OPF in pandapower. There is a tutorial how to do it.

lalala215, 2022-08-31:
OK！ Thanks a lot.
I think add the ramp rate constraint in the max_p_mw and min_p_mw is a good way, since I only consider one time step OPF problem.

closed 2022-08-31 by lalala215