e2nIEE/pandapower #1695: External Grid defined but pp still leads to user warning and non converging NR

AnkurArohi, 2022-08-31:
pp.runpp(net)

    _powerflow(net, **kwargs)
  File "/home/ankur/emsconda/envs/netztopologie/lib/python3.9/site-packages/pandapower/powerflow.py", line 75, in _powerflow
    ppc, ppci = _pd2ppc(net)
  File "/home/ankur/emsconda/envs/netztopologie/lib/python3.9/site-packages/pandapower/pd2ppc.py", line 153, in _pd2ppc
    _check_for_reference_bus(ppc)
  File "/home/ankur/emsconda/envs/netztopologie/lib/python3.9/site-packages/pandapower/build_gen.py", line 322, in _check_for_reference_bus
    raise UserWarning("No reference bus is available. Either add an ext_grid or a gen with slack=True")

UserWarning: No reference bus is available. Either add an ext_grid or a gen with slack=True
net.ext_grid
Out[114]: 
      name   bus    vm_pu  va_degree  slack_weight  in_service  topo_island
0  no_type  3104  1.07375  -2.904865           1.0        True          NaN

rbolgaryn, 2022-08-31:
Hi @AnkurArohi ,

you can use the pp.diagnostic function, maybe it will give more info.

It is possible that the ext_grid is disconnected from the rest of the grid by a switch or similar

AnkurArohi, 2022-08-31:
Thanks @rbolgaryn 

Yes but the thing is i also used pp.drop_inactive_elements()

Still the bus to which the ext_grid is connected as well as the ext_grid are still in the network

AnkurArohi, 2022-08-31:
_____________ PANDAPOWER DIAGNOSTIC TOOL _____________ 
 --------
 --------
 --------
Checking for impedance values close to zero...
impedance 2: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 9: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 10: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 23: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 26: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 27: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 41: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 43: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 49: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 67: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 68: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 73: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 81: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 82: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 84: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 92: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 99: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 100: r_pu <= 1e-05 or x_pu <= 1e-05
impedance 104: r_pu <= 1e-05 or x_pu <= 1e-05
Power flow still does not converge after replacing implausible elements with switches.
SUMMARY: 19 element(s) with impedance values close to zero found.
 --------
 --------
 --------
Check for overload failed due to the following error:
No reference bus is available. Either add an ext_grid or a gen with slack=True
 --------
 --------
Checking switch configuration...
Power flow still does not converge with all switches closed.
 --------
 --------
 --------
 --------
numba_comparison failed due to the following error:
No reference bus is available. Either add an ext_grid or a gen with slack=True
 --------
 --------
_____________ END OF PANDAPOWER DIAGNOSTIC _____________ 
Out[117]: 
{'impedance_values_close_to_zero': [{'impedance': [2,
    9,
    10,
    23,
    26,
    27,
    41,
    43,
    49,
    67,
    68,
    73,
    81,
    82,
    84,
    92,
    99,
    100,
    104]},
  {'loadflow_converges_with_switch_replacement': False}],
 'wrong_switch_configuration': False}

AnkurArohi, 2022-08-31:
I found that the bus to which the ext_grid was connected was bymistake set out of service , sorry, I set it back to service

AnkurArohi, 2022-08-31:
The powerflow still does not converge.

AnkurArohi, 2022-08-31:
Moving forward i tried to increase both the rft_pu and xft_pu for the impedances as below

diag["impedance_values_close_to_zero"][0]["impedance"]
Out[152]: [2, 9, 10, 23, 26, 27, 41, 43, 49, 67, 68, 73, 81, 82, 84, 92, 99, 100, 104]
n.impedance.loc[diag["impedance_values_close_to_zero"][0]["impedance"],"xft_pu"]=0.1
n.impedance.loc[diag["impedance_values_close_to_zero"][0]["impedance"],"rft_pu"]=0.1
pp.diagnostic(net)

Still it gives the same impedances as output and with the same message that the values are very low

AnkurArohi, 2022-08-31:
Although I found these extra warnings

/home/ankur/emsconda/envs/netztopologie/lib/python3.9/site-packages/scipy/sparse/linalg/_dsolve/linsolve.py:230: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
/home/ankur/emsconda/envs/netztopologie/lib/python3.9/site-packages/pandapower/pf/create_jacobian.py:18: RuntimeWarning: invalid value encountered in true_divide
  dVm_x, dVa_x = dSbus_dV_numba_sparse(Ybus.data, Ybus.indptr, Ybus.indices, V, V / abs(V), Ibus)

AnkurArohi, 2022-08-31:
The ext_grid is connected to the main grid

AnkurArohi, 2022-09-01:
Hi @rbolgaryn 

I have debugged further and this is how the Jacobian matrix looks which is tried to be inverted using spsolve.gssv() function, which further leads to the warning that the Matrix is singular and cant be inverted

[J.csv](https://github.com/e2nIEE/pandapower/files/9467956/J.csv)
[Ybus.csv](https://github.com/e2nIEE/pandapower/files/9467957/Ybus.csv)

rbolgaryn, 2022-09-03:
> Moving forward i tried to increase both the rft_pu and xft_pu for the impedances as below
>
>
>
> diag["impedance_values_close_to_zero"][0]["impedance"]
>
> Out[152]: [2, 9, 10, 23, 26, 27, 41, 43, 49, 67, [...]

If the values are still close to zero, it means somehow it didn't work to change the impedance values. You should achieve the impedance values not close to zero.

Roman

AnkurArohi, 2022-09-05:
Hi @rbolgaryn 
thanks,
So, the impedance values are currectl set I had ignored the xtf values while increasing theimpedance now this error is no more, 
But the PP still fails to converge, the external grid is corrrectly connected to the main topological island where i am hoping a converging NR status.

rbolgaryn, 2022-09-05:
sorry can't help you any further with this.
you should keep debugging the grid.
you can run dc powerflow and take a look at voltage values and loading values.

closed 2022-09-06 by AnkurArohi