e2nIEE/pandapower #1704: problem running runpp

user960507, 2022-09-04:
I am trying to run an example (opf_basic.ipynb) from the tutorial folder, and just got this output when running
TypeError:  expected dtype object, got numpy.dtype[float64]

Trace came from the file build_branch.py line 563

  The same exception is raised when running another example involving powerflow. I just want to know if it is a missconfiguration on my install or a known bug

Thanks in advance

SteffenMeinecke, 2022-09-05:
Hi @user960507 ,
thanks for raising this issue.
It seems to be a problem due to not matching python package version. Which versions are installed (python, pandapower, numpy and pandas)?

user960507, 2022-09-06:
Dear Steffen Meinecke:

Thanks for the attention on the request.

This issue started when i updated pandapower package (using pypi) last saturday, before that, i was using the version 2.4 and was perfectly running.

Currently i am using:
python 3.7.4
pandas 1.3.5
numpy 1.21.6
pandapower 2.10.1

of those only python is a bit outdated, i am going to try updating it to a higher version (maybe py 3.9??) and i will keep you up to date with the progress.

Thanks in advance for your help

Abel

Sent with [Proton Mail](https://proton.me/) secure email.

AnkurArohi, 2022-09-06:
@user960507 

This is the error due to wrong dtype, you need to find which parameter has wrong dtype.

I had this issue with slack_weight factor  dtype.

Maybe you could test the dtypes of the attributes.

user960507, 2022-09-07:
hiiii

Feedback on the issue. Problem solved!!!...It seems to be a package compatibility problem. I have upgraded the env and all was fine:

python 3.10.6
pandas 1.4.4
numpy 1.23.2
pandapower 2.10.1

The example i was trying was "powerflow.ipynb" in the tutorial directory, after the upgrade runpp function worked fine.

Thank you very much for your help.

Abel A

Sent with [Proton Mail](https://proton.me/) secure email.

SteffenMeinecke, 2022-09-07:
Perfect, @user960507 thanks for reporting.

closed 2022-09-07 by SteffenMeinecke