e2nIEE/pandapower #1722: Modeling switches in network

hrs038, 2022-09-29:
I was begining to use pandapower for modelling analysis purpose. The system is a  low voltage network (0.4 kV) with two grid feed source possibilities. It contains 4 buses and the between bus 2 and 3, I plan to insert switch for connect or disconnect. In case of disconnection, it will work as two separated systems with bus 1 and 4 as slack buses for each network separately. However, I came across a problem in the power flow when I am trying to use two cases with an "open" and "close" line. The issue is, even I use the "switch" to disconnect a line from the network, it shows some power flow through that line (line 2-3). If I remove (comment out in the code) the line, then it shows the result as if the line is out  (very very small power flow as shown in the minimal example of pandapower). It would be a great help if someone can provide some info. Thanks.

rbolgaryn, 2022-10-05:
Hi @hrs038 ,

can you provide the code of your example?
Are you sure that it is not the case of "loading current" of the line?

Best regards,
Roman

hrs038, 2022-10-05:
Hi,
The issue is resolved.

Thank you very much.

Best Regards

import pandapower as pp

net = pp.create_empty_network()
b1 = pp.create_bus(net, vn_kv=0.4, name="b1", index=1)
b2 = pp.create_bus(net, vn_kv=0.4, name="b2", index=2)
b3 = pp.create_bus(net, vn_kv=0.4, name="b3", index=3)
b4 = pp.create_bus(net, vn_kv=0.4, name="b4", index=4)
#create bus elements
pp.create_ext_grid(net, bus=1, vm_pu=1.0, name="ST1")
pp.create_ext_grid(net, bus=4, vm_pu=1.0, name="ST2")
#pp.create_sgen(net, bus=4, p_mw = 0.2)
#create bus elements
pp.create_load(net, bus=1, p_mw=0.3, q_mvar=0.006, name="Load1")
pp.create_load(net, bus=2, p_mw=0.3, q_mvar=0.006, name="Load2")
pp.create_load(net, bus=3, p_mw=0.15, q_mvar=0.006, name="Load3")
pp.create_load(net, bus=4, p_mw=0.15, q_mvar=0.006, name="Load4")

#create branch elements
line_1= {"r_ohm_per_km": 0.0075, "x_ohm_per_km": 0.00033936, "c_nf_per_km": 1, "max_i_ka": 3.0, "type": "cs"}
line_2= {"r_ohm_per_km": 0.0125, "x_ohm_per_km": 0.00033936, "c_nf_per_km": 1, "max_i_ka": 3.0, "type": "cs"}
line_3= {"r_ohm_per_km": 100000000, "x_ohm_per_km": 100000000, "c_nf_per_km": 1, "max_i_ka": 3.0, "type": "cs"}
pp.create_std_type(net, name="line_1", data=line_1, element="line")
pp.create_std_type(net, name="line_2", data=line_2, element="line")
pp.create_std_type(net, name="line_3", data=line_3, element="line")
pp.available_std_types(net, element="line")

pp.create_line(net, index =1, from_bus=1, to_bus=2, length_km=1, name="Line12",std_type="line_1", in_service="True")
#pp.create_line(net, from_bus=b2, to_bus=b3, length_km=1, name="Line12", r_ohm_per_km="0.125", x_ohm_per_km="0.02121", c_nf_per_km="210", max_loading_percent="100")
pp.create_line(net, index =2, from_bus=2, to_bus=3, length_km=1, name="Line23",std_type="line_2", in_service="True")
pp.create_line(net, index =3, from_bus=3, to_bus=4, length_km=1, name="Line34",std_type="line_1", in_service="True")
pp.create_switch(net, bus=2, element=2, et="l", type ="CB", z_ohm = 0, closed="False")
pp.create_switch(net, bus=3, element=2, et="l", type ="CB", z_ohm = 0, closed="False")
pp.runpp(net)
print(net.res_bus)
print(net.res_line)

rbolgaryn, 2022-10-06:
You are welcome!

closed 2022-10-06 by rbolgaryn