e2nIEE/pandapower #1726: OPF does not converge with standard Testcases

Christian-95, 2022-10-05:
Hello, 

i tried to test the opf functionalities in Pandapower a few days.
I used Power System Test Cases case9 case39 case118 and case3120sp for example, to run an opf with the given functions and solverconverter in PYPOWER and POWERMODELS.jl.
PYPOWER opf does not converge with bigger cases like case2848rte or case3120sp for example (also not with a DCOPF).
But as i want to use a mixed integer linear programming solver i want to focus on POWERMODELS.jl that has functionalities that support this.
**To focus the problem**: I use the runpm_dc_opf Method for Powermodels.
`net = pp.networks.case5()`
`pp.runpm_dc_opf(net, correct_pm_network_data=False,silence=False)`
--> leads to a successfull opf
`pp.runpm_dc_opf(net,silence=False)` -means that `correct_pm_network_data `standard value is `True`
--> leads to "corrections":

> [info | PowerModels]: removing 1 cost terms from generator 4: Any[10.0, 0.0]
> [info | PowerModels]: removing 1 cost terms from generator 1: Any[40.0, 0.0]
> [info | PowerModels]: removing 1 cost terms from generator 5: Any[15.0, 0.0]
> [info | PowerModels]: removing 1 cost terms from generator 2: Any[14.0, 0.0]
> [info | PowerModels]: removing 1 cost terms from generator 3: Any[30.0, 0.0]

and did not converge:

>  raise OPFNotConverged("PowerModels.jl OPF not converged")
> pandapower.optimal_powerflow.OPFNotConverged: PowerModels.jl OPF not converged

So firstly we must note: the "auto detect network correction" leads to a case that prevent the opf to converge.

Next: case39() does not converge at all with Powermodels (but in Pypower):
`net = pp.networks.case39()`
`pp.runpm_dc_opf(net,silence=False)`
--> OPF not converged

Now i use juniper as main solver for case5 and case39:
`net = pp.networks.case5()`
`pp.runpm_dc_opf(net, correct_pm_network_data=False,silence=False, pm_solver="juniper")`
--> leads to a successfull opf

`net = pp.networks.case39()`
`pp.runpm_dc_opf(net, silence=False, pm_solver="juniper")`

> [warn | InfrastructureModels]: model has no results, solution cannot be built
> Traceback (most recent call last):
>   File "C:\Users\nsobisch\Desktop\PCCM2.0\pccm2.0\pandamodelstest.py", line 83, in <module>
>     pp.runpm_dc_opf(net, silence=False, pm_solver="juniper")
>   File "C:\Users\nsobisch\AppData\Local\Programs\Python\Python310\lib\site-packages\pandapower\runpm.py", line 112, in runpm_dc_opf
>     _runpm(net, delete_buffer_file=delete_buffer_file, pm_file_path=pm_file_path, pdm_dev_mode=pdm_dev_mode)
>   File "C:\Users\nsobisch\AppData\Local\Programs\Python\Python310\lib\site-packages\pandapower\opf\run_powermodels.py", line 52, in _runpm
>     raise e
>   File "C:\Users\nsobisch\AppData\Local\Programs\Python\Python310\lib\site-packages\pandapower\opf\run_powermodels.py", line 41, in _runpm
>     read_pm_results_to_net(net, ppc, ppci, result_pm)
>   File "C:\Users\nsobisch\AppData\Local\Programs\Python\Python310\lib\site-packages\pandapower\converter\powermodels\from_pm.py", line 19, in read_pm_results_to_net
>     result, multinetwork = pm_results_to_ppc_results(net, ppc, ppci, result_pm)
>   File "C:\Users\nsobisch\AppData\Local\Programs\Python\Python310\lib\site-packages\pandapower\converter\powermodels\from_pm.py", line 89, in pm_results_to_ppc_results
>     return result, multinetwork
> UnboundLocalError: local variable 'result' referenced before assignment

I have to say that all tested networks converged in matpower easily, i also used the mpc to pandapower converter, to be sure that the models in the pp.networks lib are not broken, and converted the working networks from matpower to pandapower, same results.

I hope you can find the error for that.

Greetings
Christian

Christian-95, 2022-10-09:
Yes that works. :) I tried Powermodels DCOPF and ACOPF, with case39 both works, with case3120sp only DC converged. 
Can you tell me why a sn_mva of 1 works and is it correct? The descriptions of case39 for example says it should be 100 MVA. 
Do all networks have to be to set on 1 sn_mva to get a working opf with Powermodels, or which way it depends on that?

Greetings
Christians

rbolgaryn, 2022-10-11:
It seems like for now it is only implemented for sn_mva = 1

But we should adjust that so that sn_mva != 1 is supported.

Roman

closed 2022-10-18 by ZhengLiu1119