e2nIEE/pandapower #1732: Changing data type of net.load.controllable after saving to and loading from sqlite

DavidEnergie, 2022-10-18:
Hi,

recently having upated from pp 2.4 to pp 2.9, I ran into an issue saving and loading network to SQL.
When I save a large network, that has been converted from MPC, using pp.to_sqlite(net,file_ppc +'.sqlite', include_results=False) and later loading the network again using pp.from_sqlite(file_ppc), a data type changes.

The data type of net.load.controllable changes from 'bool', containing 

> FALSE

 (i.e.), to 'object', containing 0 as a string

>  '0'

```
net.load.controllable.unique()
array([0], dtype=object)
```

Did anyone experience similar issues and could give me a pointer on the origin and possible mitigations of this issue?

rbolgaryn, 2022-10-19:
Hi @DavidEnergie ,

can you please provide a minimal example for this issue as a failing test?

Best regards, 
Roman

rbolgaryn, 2023-07-19:
Dear @DavidEnergie ,

please provide a minimal test example.

DavidEnergie, 2023-08-27:
Sorry I had forgotten about this issue. I will look into it in the next week, trying to reproduce the error.

KS-HTK, 2024-11-22:
@DavidEnergie,
is this still an Issue or can this be closed? If the issue still persists can you please provide an example?