e2nIEE/pandapower #1740: Discrepancies in OPF solutions when calling PowerModels.jl via PandaPower versus calling PowerModels.jl directly

ElPiloto, 2022-10-28:
Hello,

I'm trying to solve the OPF problems in the [pglib](https://github.com/power-grid-lib/pglib-opf) repo, but I'm encountering different results using pandapower versus powermodels.jl.  Many cases simply don't converge, other cases provide different objective values. To be clear, I'm always using PowerModels.jl to solve, but the difference is whether or not I call it from python via `pandapower.runpm_ac_opf` versus directly in julia via `PowerModels.solve_ac_opf`.

Below I've included some some specific code snippets which illustrate the difference and also additional information.

## Calling OPF Solver in PowerModels.jl via PandaPower
```
import pandapower as pp
improt pandapower.converter as converter

case_file = "/data/pglib_opf_case300_ieee.m"
net = converter.from_mpc(case_file)

pp.runpm_ac_opf(net, pm_log_level=5)

## Fails to converge after 542 iterations
```

## Calling OPF Solver in PowerModels.jl via Julia script
```
using PowerModels
using Ipopt

case_file = "/data/pglib_opf_case300_ieee.m"
solve_ac_opf(case_file)

## Converges after 33 iterations
```

# Additional Information:
* I have seen that some pglib-opf cases require setting `sn_mva = 1.` (or 100.) in order to converge - even with this snippet added it turns out that we get different answers PowerModels and PandaPower.
* It looks like the discrepancy in solutions arises from a discrepancy between the problem that is being sent to Ipopt.  I've attached screenshots of the Ipopt output from PowerModels.jl and Pandapower for the 3 bus case (`pglib_opf_case3_lmbd`).  You can see that there are a different number of variables.
[image]
[image]

* I am using the following versions:
  - python 3.9.2
  - pandapower: v2.10.1 (initially installed via pip and then subsequently tried to update the "site-package" in my virtual env  by grabbing the latest version from github, but not 100% this worked - maybe it's still using the old version somehow)
  
  - Julia 1.5.0
  - Ipopt.jl v0.9.1
  - PowerModels.jl v0.19.2
  - PandaModels v0.7.0 (initially installed via Pkg but then I tried to manually update to the latest version from Github dev branch, not sure if this worked or if it was still using an older version)
  
  - pglib-opf (v21.07 - cloned at [commit](01681386d084d8bd03b429abcd1ee6966f68b9a3))

ZhengLiu1119, 2023-07-21:
hi @ElPiloto,

I apologize for the late response. Thank you very much for providing the code and raising the question. You are right; there is indeed a convergence issue after testing this power grid. I believe the problem might lie here `net = converter.from_mpc(case_file)` because after converting the network, the basic power flow calculation in pandapower with `pp.runpp(net)` can not be converged. I would like to focus on resolving the converter issue first. I will keep you informed if we discover anything.

ZhengLiu1119, 2023-07-21:
hi @SteffenMeinecke , 

after converting the grid, the power flow calculation with runpp is not converged, and it is generating some warning messages
"**hp.pandapower.converter.pypower.from_ppc - WARNING: There are 13 branches which are considered as trafos - due to ratio unequal 0 or 1 - but connect same voltage levels. hp.pandapower.converter.pypower.from_ppc - INFO: Transformers always behave inductive consumpting but the susceptance of pypower branches [28 29 30 31 32 33 35 36 40 41 43 46] (hv_bus, lv_bus)=([105 107 110 135 136 147 126 126 197 215 209 209], [105 107 110 135 136 147 126 126 197 215 209 209]) is positive.**"

could you please verify whether the grid has been converted correctly?

```
import pandapower as pp
impott pandapower.converter as converter
case_file = "pglib_opf_case300_ieee.m"
net = converter.from_mpc(case_file)
pp.runpp(net)
```

(you can find the grid model "pglib_opf_case300_ieee.m" [here](https://github.com/power-grid-lib/pglib-opf/blob/01681386d084d8bd03b429abcd1ee6966f68b9a3/pglib_opf_case300_ieee.m#L24))

SteffenMeinecke, 2023-07-24:
@ZhengLiu1119 The author of the converter is me. Are there any ambiguities in the warning message? As long as the information from the net is available, unambiguous and reasonable (it is not in the branches mentioned), the net will be converted correctly.

ZhengLiu1119, 2023-07-25:
Hello @ElPiloto ,

By default, in pandapower, the voltage of the slack node remains fixed, even after we set the upper and lower voltage limits for this node (e.g., `Vmax = 1.1`, `Vmin = 0.9`). To activate the `Vmax` and `Vmin` constraints, we need to set the slack (`ext_grid` in `pandapower`) to be controllable as follows:

```
import pandapower as pp
import pandapower.converter as converter

case_file = "pglib_opf_case3_lmbd.m"
net = converter.from_mpc(case_file)
net.ext_grid["controllable"] = True  # set slack as controllable
pp.runpm_ac_opf(net, pm_log_level=5)
```

I have just tested this, only for the **3 bus case**, and the results match the ones described in the .m file (**pglib_opf_case3_lmbd.m**). You can try it yourself.

However, the network with 300 nodes still fails to converge. I suspect that there might be an issue with the parameters not matching, as I noticed some differences between the converted pandapower network and the original m-file in certain parameters.

yamizi, 2024-03-18:
Hello,
I am facing the same issue with AC-OPF on the "case1354pegase". It converges well on Matpower but never converges on pandapower, even when using "runpm_ac_opf".

It was described here as [Known Problem](https://pandapower.readthedocs.io/en/develop/opf/known_problems.html). So it seems the team behind Pandapower is aware of it, the proposed solution is "branch limits"., however, even when increasing the line limits, the Vm limits... no convergence is achieved.

Did any manage to achieve OPF for  large cases in Pandapower (while they converge in PowerModels.jl) ?
best,
Salah

vogt31337, 2026-11-13:
fixed with #3330

closed 2026-11-13 by vogt31337