e2nIEE/pandapower #1748: Finding Q-potential at a fixed bus for the whole network using Jacobian matrix

AnkurArohi, 2022-11-04:
I want to find the change in reactive power at a fixed bus due to all reactive power changes in the controllable reactive power equipments in the network.

I was anaysing the Jacobian Matrix, lets say at node "n" I would like to find the reactive power potential so to speak the maximum and minimum reactive powers possible at this point at different network operational Reactive power values at different buses.

Looking at the matrix I am thinking of doing following pseudo code operations

1. Extract the Jacobian Matrix for a NR converged network.

2. Extract the bus indices where there are Reactive power controllable equioments

3. For the fixed active power determine the maximum stable voltage difference possible from the current value, considering the reactive power control (using Q(V) equation for a fixed P).
4. For previously defined del(V) both positive and negative use the jacobian ratios to find the del(Q) at the nth node

> del(Qn) pos= Jacobianfactors{all controllable nodes}*del(V)pos
> del(Qn) neg= Jacobianfactors{all controllable nodes}*del(V)neg

Do you have other better suggestions, do you think this is a doable approach, any limmitations?

Highly appreciate your reply.

rbolgaryn, 2022-11-21:
Dear @AnkurArohi ,

I think this is a good approach.
You need to make sure that your indexing and bus mapping for the inverted J are correct:

1) the pq-lookup, pvpq-lookup for internal indexing in the J
2) The location of dQ/dV is different in J^-1 from the location of dQ/dV in J

jwiemer112, 2023-01-23:
Can this issue be closed?

closed 2023-02-14 by AnkurArohi