e2nIEE/pandapower #1755: Individual converter contributions for short circuit

amit20patil, 2022-11-18:
I'm using the short circuit calculation of pandapower for a fault analysis. From the documentation, I've seen that using the calc_sc we can obtain the current flows for a type of fault. However, some things are not clear to me:

1) net.res_bus_sc supposedly gives bus currents. What is meant by bus currents?

2) line.res_bus_sc directly gives the current flows for a fault with some impedance. Is it possible to obtain the contributions of a converter specifically rather than the resulting flows?
For example, for a fault z impedance at bus 1, the converter at bus 2 supplies 0.1 kA current and converter at bus 3 supplies 0.2 kA.

rbolgaryn, 2022-12-01:
Dear @amit20patil ,

1. res_bus_sc gives the max. current caused by the short circuit at that bus
2. the contribution of a full size converter is given by $\frac{S_{sgen}}{\sqrt(3) \cdot V_n} \cdot k$ where $S_{sgen}$ is sn_mva of the full size converter (sgen), $V_n$ is the rated voltage in kV at the connected bus, and k is the ratio of short circuit current to the rated current (typically 1.3). We are not providing separate results in pandapower for the converters at this time.

rbolgaryn, 2022-12-02:
You can access the individual contributions in the ppc:

net._ppc["bus"][:, IKCV]

where IKCV is imported from pandapower.pypower.idx_bus_sc

Best regards,
Roman

amit20patil, 2022-12-06:
@rbolgaryn Thanks! I will try this out.

While reading the output from calc_sc, there are options "branch_results" and "return_all_currents". If they are "True", it seems like the bus currents for a fault at each line is printed. Is this a correct understanding of the output?

For example:
  net = nw.case33bw()
  net.ext_grid["s_sc_min_mva"] = 100
  net.ext_grid["s_sc_max_mva"] = 1000
  net.ext_grid["rx_min"] = 0.2
  net.ext_grid["rx_max"] = 0.35
  net.ext_grid["r0x0_max"] = 0.4
  net.ext_grid["x0x_max"] = 1.0

  for b, p in [(10, 100), (20, 50), (23, 50), (28, 50)]:
    pp.create_sgen(net, b, p_mw=p, sn_mva=p, k=1.3)
  # pp.create_gen(net, 2, p_mw=0, vm_pu=1.0, cos_phi=0.8, vn_kv=22, sn_mva=5, xdss_pu=0.2, rdss_ohm=0.005)
  net.line["endtemp_degree"] = 20
  net.line["r0_ohm_per_km"] = 0.244
  net.line["x0_ohm_per_km"] = 0.336
  net.line["c0_nf_per_km"] = 2000
sc.calc_sc(net, fault="1ph", topology="radial", case="max", ith=True, ip=True, branch_results=True, return_all_currents=True, r_fault_ohm=3, x_fault_ohm=3)
with pd.option_context('display.max_rows', None, 'display.max_columns', None):
    print(net.res_line_sc)

Gives the following output:

               ikss_ka         ip_ka  ith_ka
line bus                                    
0    0    1.482139e+01  2.096062e+01     NaN
     1    1.744427e+00  2.643988e+00     NaN
     ...
     31   9.664526e+00  1.361801e+01     NaN
     32   1.000328e+01  1.409646e+01     NaN
1    0    1.185711e+01  1.676849e+01     NaN
     1    1.185711e+01  1.676849e+01     NaN
     ...
     31   6.700247e+00  9.425883e+00     NaN
     32   7.039006e+00  9.904340e+00     NaN
2    0    8.892836e+00  1.257637e+01     NaN
     1    8.892836e+00  1.257637e+01     NaN
     ...
     
Does the line column mean the fault is induced on that line and bus column indicate the index of the bus for which the current values are printed?

rbolgaryn, 2022-12-07:
Dear @amit20patil ,

the faults occur at the buses. 
The line results for all lines return the short-circuit current values in the lines caused by the fault at the respective bus.

Roman