e2nIEE/pandapower #1766: The first power flow calc costs more time than others

wbxsos, 2022-12-02:
In pandapower 2.7.0, when doing a cycle calculation like **runpp(net, algorithm = 'bfsm')**， I got a confusing issue, the first calc is much slower than others.
At first, I thought that connected to the cycle work because I just updated the measure data of each element after one calculation was done.
Then I run a test, I generated 4 different pandapower nets named from A to D in the same thread one by one, each net calced just ones. First from A to D, the net A costed about 1000ms the others 10ms, then from D to A, the net D costed about 1000ms and others 10ms. 
Is there some preparatory work in the pandapower package before I run the "runpp" function and if so,what can I do to solve this

ascheidl, 2022-12-02:
The reason for this effect is the just-in-time-compiler numba.

For some reason I cannot recall someone decided to switch off caching for numba.

Consequently, at the first run always all numba accellerated functions are recompiled - and this takes some time.

wbxsos, 2022-12-02:
@ascheidl Thank you for your help and wish you a good day.

closed 2022-12-02 by wbxsos