e2nIEE/pandapower #1775: Bug of plot: wrong direction

DavidLudl, 2022-12-09:
Hello. 

When I using simple_plot and simple_plotly, I find that the output figure is kind of upside down. For example, this is the result of simple plot:
[image]
However, the correct picture should be like this:
[image]
My bus_geodata input format is (lat, lon), where positive lat means Northern Hemisphere and negative is Southern Hemisphere.
It seems the direction is upside down plus horizon flip compared with  map. I don't know if this is error or I should use different geodata input format?

Thank you for your time,
Best regards,

David

jkisse, 2022-12-16:
Hi,

it seems to me like the `x` and `y` values are reversed. Please try it with longitude values in the `net.bus_geodata.x` column and latitude as `net.bus_geodata.y`. "lat" and "lon" as column names might be misunderstood by the plotting function.

DavidLudl, 2022-12-16:
Hi,

I tried this method before, by input (lon, lat) format, which is similar with your suggestion. However, another error was rised which said x should with in -90 to 90. Therefore, I suppose net.bus_geodata.x allow only latitude.

Here is the screen shoot:
[image]

jkisse, 2022-12-16:
I tried to reproduce the issue with the following example and `simple_plot` worked, even with x > 90. Could you please provide a minimal example for your case?
```
import pandapower
import pandapower.networks as pn
net = pn.GBnetwork()
net.bus_geodata.x += 90
print(net.bus_geodata.max())
pandapower.plotting.simple_plot(net)
```
pandapower 2.11.0
geopy 2.2.0
Python 3.9.9

DavidLudl, 2022-12-16:
Hello, 

Thank you for your reply.

I tried with these expressions as yours:

"x = net.bus_geodata.x
y = net.bus_geodata.y
net.bus_geodata.x = y
net.bus_geodata.y = x

plot.simple_plot(net, show_plot=True)"

Then I get the correct output:
[image]

Thank you for your help.

closed 2022-12-16 by jkisse