e2nIEE/pandapower #1776: BUG: switch in service  in  pp.topology.create_nxgraph

dyzxtcwl, 2022-12-11:
**when i runed pp.topology.create_nxgraph , it turned out that:**

  File "d:\anaconda3\envs\spyder\lib\site-packages\pandapower\topology\create_graph.py", line 246, in create_nxgraph
    indices, parameter = init_par(switch, calc_branch_impedances)

ValueError: too many values to unpack (expected 2)

**Then I found in init_par() it was that :**
indices, parameter = init_par(switch, calc_branch_impedances)

def init_par(tab, calc_branch_impedances=False):
      if "in_service" in tab:
        return indices, parameters, **tab.in_service.values.copy()**
    else:
        return indices, parameters

**So I try to modified it to:**
indices, parameter = init_par(switch.drop('in_service', axis=1), calc_branch_impedances)

jkisse, 2022-12-16:
Hi,
`in_service` is not a default column for switches and will probably be nowhere considered in the power flow. Only `closed` is considered.
It would be interesting to know if there is a loophole somewhere that creates in_service columns for switches.
Do you use any of the networks from the `pandapower.networks` package? If not, did you create the switches by the `create_switch` function?

Apart from that, your fix seems reasonable. I just wonder if an Error/Warning should be raised, because it will likely lead to unexpected behavior if the user thinks `net.switch.in_service` will be considered by the power flow calculation.

dyzxtcwl, 2023-06-17:
> Hi, `in_service` is not a default column for switches and will probably be nowhere considered in the power flow. Only `closed` is considered. It would be interesting to know if there is a loophole [...]

There was an another problem with "in_service" column of switches in the funtion "pandapower.drop_inactive_elements()".
This funtion needs the "in_service" column of switches otherwise it will raise an Error.

Jhei1195, 2024-02-13:
Hello,
I have the same issue with pandapower version 2.13. Is there a proper solution now?

pawellytaev, 2024-11-25:
@Jhei1195 @dyzxtcwl do you have a minimal example code to show the issue?

vogt31337, 2025-03-06:
short example which breaks the code.
```
import pandapower as pp
import pandapower.networks as nw
from pandapower.topology import create_nxgraph

net = nw.mv_oberrhein()
net.switch['in_service'] = True
pp.runpp(net)
mg = create_nxgraph(net)
print("finished.")
```

Problem is in create_graph.py, the function init_par returns either two or three values and this seems to be the problem.

closed 2025-03-07 by vogt31337