e2nIEE/pandapower #1783: opf does not converge - mv oberrhein

catarinapmartins, 2022-12-14:
Hello everyone! 
I'm trying to run an opf with the mv oberrhein grid that is defined in pandapower but I can't understand why it is not converging.

I have tried to run the opf with costs or without costs, and with the generators and loads being controllable or not controllable but it does not converge in any of the cases. Besides I also tried to change the initialization and the detla factor in the opf but still doensn't converge.

Below I leave my code:

-------------------

mv_oberrhein = pp.networks.mv_oberrhein(scenario="generation") 
max_loading_percent_trafo = 120
max_loading_percent_line =110
max_vm_pu = 1.1
min_vm_pu = 0.9

mv_oberrhein.load["sn_mva"] = [1]*len(mv_oberrhein.load)
mv_oberrhein.line["max_loading_percent"] = [max_loading_percent_line]*(len(mv_oberrhein.line))
mv_oberrhein.bus["max_vm_pu"] = [max_vm_pu]*len(mv_oberrhein.bus)
mv_oberrhein.bus["min_vm_pu"] = [min_vm_pu]*len(mv_oberrhein.bus)
mv_oberrhein.trafo["max_loading_percent"] = [max_loading_percent_trafo]*len(mv_oberrhein.trafo)

mv_oberrhein.sgen["controllable"]=[True]*len(mv_oberrhein.sgen)
mv_oberrhein.sgen["min_p_mw"] = [0]*len(mv_oberrhein.sgen)
mv_oberrhein.sgen["max_p_mw"] = mv_oberrhein.sgen["p_mw"]
mv_oberrhein.sgen["min_q_mvar"] = [0]*len(mv_oberrhein.sgen)
mv_oberrhein.sgen["max_q_mvar"] = [0]*len(mv_oberrhein.sgen)

mv_oberrhein.load["controllable"]=[False]*len(mv_oberrhein.load)
mv_oberrhein.load["min_p_mw"] = [0]*len(mv_oberrhein.load)
mv_oberrhein.load["max_p_mw"] = mv_oberrhein.load["p_mw"]
mv_oberrhein.load["min_q_mvar"] = [0]*len(mv_oberrhein.load)
mv_oberrhein.load["max_q_mvar"] = mv_oberrhein.load["q_mvar"]

for gen in range(len(mv_oberrhein.sgen)):
    pp.create_poly_cost(net=mv_oberrhein, element=mv_oberrhein.sgen.index[gen] , et='sgen', cp1_eur_per_mw=-1)
    
pp.runopp(mv_oberrhein)

------------

I had a look into the issues here in git, but none of them has helped, and the examples by pandapower are all with very simple grids. 

I'm not sure if I'm forgetting any parameter or doing something wrong, any help is really appreciated!

jwiemer112, 2022-12-23:
Hey, the OPF is a bit tricky. You can use the `verbose=True` parameter to get further information. 

If you set `calculates_voltage_angles = False` the OPF converges. In meshed grids voltage_anlges should be calculated, but Oberrhein is operated in a radial manner so it is ok to set it to false (see: https://github.com/e2nIEE/pandapower/issues/489)

Here is my code snippet for you. It works fine and also demonstrates the curtailment by injecting so much power that the sgens
have to be curtailed in order to match the constraints. 

`import pandapower as pp
from pandapower.networks import mv_oberrhein

net = mv_oberrhein(scenario="generation")

max_loading_percent_trafo = 120
max_loading_percent_line = 110

max_vm_pu = 1.1
min_vm_pu = 0.9

net.load["sn_mva"] = 1

net.line["max_loading_percent"] = max_loading_percent_line

net.bus["max_vm_pu"] = max_vm_pu
net.bus["min_vm_pu"] = min_vm_pu

net.trafo["max_loading_percent"] = max_loading_percent_trafo

net.sgen.scaling = 3

net.sgen["controllable"] = True
net.sgen["min_p_mw"] = 0
net.sgen["max_p_mw"] = net.sgen["p_mw"] * net.sgen.scaling
net.sgen["min_q_mvar"] = 0
net.sgen["max_q_mvar"] = 0

net.load["controllable"] = False

for sgen_idx in net.sgen.index:
    pp.create_poly_cost(net=net, element=sgen_idx, et='sgen', cp1_eur_per_mw=-1)

pp.runpp(net)

line_max = net.res_line.loading_percent.max()
trafo_max = net.res_trafo.loading_percent.max()
bus_max = net.res_bus.vm_pu.max()
bus_min = net.res_bus.vm_pu.min()

print(f"Max. Line Loading [%]: {line_max}, Max. Trafo Loading [%]: {trafo_max}, Max. Voltage [p.u.]: "
      f"{bus_max}, Min. Voltage [p.u.]: {bus_min}")

pp.runopp(net, verbose=False, calculate_voltage_angles=False)

line_max = net.res_line.loading_percent.max()
trafo_max = net.res_trafo.loading_percent.max()
bus_max = net.res_bus.vm_pu.max()
bus_min = net.res_bus.vm_pu.min()

print(f"Max. Line Loading [%]: {line_max}, Max. Trafo Loading [%]: {trafo_max}, Max. Voltage [p.u.]: "
      f"{bus_max}, Min. Voltage [p.u.]: {bus_min}")
`

Greetz and happy christmas holidays
Jan

catarinapmartins, 2022-12-23:
Yes, I was using verbose=True but I wasn't being able to get the convergence problem.

But with that option it does converge, thank you so much!!

Happy holidays! :)

closed 2022-12-25 by jwiemer112