e2nIEE/pandapower #1785: plotly | weighted  marker trace

jwiemer112, 2022-12-14:
The new feature to create weighted plots which respect the generation "doesn't work" when you plot on a map and transform
the geocoords in `simple_plotly()`. The `create_weighted_marker_trace ` doesnt consider the projection that is passed to simple_plotly(), in the end the net is transformed but the weighted traces are not.

A quick fix is to transform the net before plotting. In perspective, the weighted_traces should also take into account the projection.

Problem code snippet:

```
    net = mv_oberrhein()
    net.load.scaling, net.sgen.scaling = 1, 1
    # different markers and sizemodes as examples
    markers_load = create_weighted_marker_trace(net, elm_type="load", color="red",
                                                patch_type="circle", sizemode="area",
                                                marker_scaling=100)
    markers_sgen = create_weighted_marker_trace(net, elm_type="sgen", color="green",
                                                patch_type="circle", sizemode="diameter",
                                                marker_scaling=100)

    simple_plotly(net, bus_size=1, on_map=True, projection="epsg:31467", additional_traces=[markers_load, markers_sgen])

```

Quick Fix code snippet:

```
    net = mv_oberrhein()

    # transformation before plotting
    geo_data_to_latlong(net, projection="epsg:31467")

    net.load.scaling, net.sgen.scaling = 1, 1

    # different markers and sizemodes as examples
    markers_load = create_weighted_marker_trace(net, elm_type="load", color="red",
                                                patch_type="circle", sizemode="area",
                                                marker_scaling=100)
    markers_sgen = create_weighted_marker_trace(net, elm_type="sgen", color="green",
                                                patch_type="circle", sizemode="diameter",
                                                marker_scaling=100)

    simple_plotly(net, bus_size=1, on_map=True, additional_traces=[markers_load, markers_sgen])

```

rbolgaryn, 2023-07-19:
what is the status of the issue @jwiemer112 ?

KS-HTK, 2024-11-29:
I think this Issue is fixed by geodata in geojson requiring the use of WGS84 and projection now not being used in simple_plotly.

I will however add a PR for a deprecation warning if projection keyword is passed to simple_plotly.

closed 2024-11-29 by KS-HTK