e2nIEE/pandapower #1791: pandapower change log - entries missing

jkisse, 2022-12-20:
### Discussed in https://github.com/e2nIEE/pandapower/discussions/1757

<div type='discussions-op-text'>

<sup>Originally posted by **MacsBO** November 21, 2022</sup>
Hi everyone and thank you for the great work you are doing.
I think some changes are missing in the change log of the new version: for example, create_sgen now has some new parameters (generator_type, max_ik_ka, kappa, lrc_pu).
unfortunately, we noticed it because, for our needs, we had added the same "generator_type" parameter (of type integrer instead of string) to the dataframe "sgen" and after updating to the latest version all the calculations were in error...
Can you update the change log?
thanks in advance
Massimo</div>

KS-HTK, 2024-11-22:
I think it might be a bit difficult to verify now what changes had been done in past commits.

It might be better to implement a precommit check that fails if the changelog has not been modified. This would also require an option to explicitly skip this if no changelog relevant changes have been made.

jkisse, 2024-11-22:
On that note: it should be discussed whether automated release notes / change logs based on the merged Pull Requests would make sense. If the Pull request titles are well-edited and PRs are systematically tagged, it would make things easier. For example: https://docs.github.com/en/repositories/releasing-projects-on-github/automatically-generated-release-notes, PyPSA: https://github.com/PyPSA/PyPSA/releases/tag/v0.31.0