e2nIEE/pandapower #1811: Use value calculated by a Constcontroller in a other controller for timeseries calculations

ThomasL95, 2023-01-08:
Hello everyone, 

I'm trying to calculate the reactive power of a load thanks to the **relative profile** of the household dataset of Simbench. I create the load as : `load_b2 = pp.create_asymmetric_load(net,bus=LV_bus_2,p_a_mw=0.007,name="Load Bus 2")` 
then i'm controlling the active power value in the timeserie with : 
`c1 = ConstControl(net, element='asymmetric_load', variable='p_a_mw', element_index=[load_b2],
            data_source=dsL, profile_name=["H0-A_pload"], scale_factor=net.asymmetric_load.loc[load_b2].p_a_mw)`
and i want to use this calculated value in a other constant controller for calculate the reactive power value thanks to the relation **Q=P*tan(\phi)**. I think it should be possible to do it via a cascade loop control, but I don't see concretely how to realize it or to use the value calculated in the scale factor of the reactive power controller. Something like that but it doesn't work `ConstControl(net, element='asymmetric_load', variable='q_a_mvar', element_index=[load_b2],
         data_source=dsL, profile_name=["H0-A_tanphi"], scale_factor=c1.values)`. Thank you for your help

rbolgaryn, 2023-01-09:
Hi @ThomasL95 ,

I think it is easier to create another ConstControl with the same data_source and its own scaling factor for Q

jwiemer112, 2023-01-09:
Hey, if the first `ConstControl` only scales an H0 profile, I would pre calculate the whole H0 timeseries for your loads and pass it to the ConstControl. With the result of the pre-calculation I would then calculate the Q timeseries (Q=P*tan(\phi) for all values) and pass it to a second `ConstControl`.

In the end you have two ConstControls, one for P and one for Q, with pre-calculated P and Q timeseries'

Greetz Jan

jwiemer112, 2023-01-13:
Hey @ThomasL95 feel free to reopen this Issue. I think the advices are sufficient to complete your task.

Greetz Jan

closed 2023-01-13 by jwiemer112