e2nIEE/pandapower #1822: test estimation: pytest.skipif for Python version 3.7

rbolgaryn, 2023-01-11:
### Issue Report Checklist

* [ x ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [ x ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [ x ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

## Labels (if access available)
* [ x ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

jkisse, 2023-01-17:
failures are apparently not limited to python 3.7 
in [this job](https://github.com/e2nIEE/pandapower/actions/runs/3941770119/jobs/6744599286), it fails with Python 3.10

jkisse, 2023-02-03:
What is the status of this issue and the progress on fixing it?
The problem is often blocking the testing/integration pipeline.  Do you need support?

marceldipp, 2023-02-20:
With the change of the default optimization method the error could be fixed, so that test_ql_qc now runs without failures in the gitlab test environment regardless of the processor performance. I think the issue can be closed.

closed 2023-07-19 by rbolgaryn