e2nIEE/pandapower #1834: pip install pandapower does not update the converter.from_mpc() function

KishanGitASU, 2023-01-22:
### Issue Report Checklist

* [check ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [ check] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [check ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

--------------

I tried to convert a .m and load it as pandapower network. It was trying to use "loadmat" and threw error.

The local functions on my laptop are :
def from_mpc(mpc_file, f_hz=50, casename_mpc_file='mpc', validate_conversion=False):
    """
    This function converts a matpower case file (.mat) version 2 to a pandapower net.

    Note: python is 0-based while Matlab is 1-based.

    INPUT:

        **mpc_file** - path to a matpower case file (.mat).

    OPTIONAL:

        **f_hz** (int, 50) - The frequency of the network.

        **casename_mpc_file** (str, 'mpc') - The name of the variable in .mat file which contain the matpower case structure, i.e. the arrays "gen", "branch" and "bus".

    OUTPUT:

        **net** - The pandapower network

    EXAMPLE:

        import pandapower.converter as pc

        pp_net = cv.from_mpc('case9.mat', f_hz=60)

    """
    ppc = _mpc2ppc(mpc_file, casename_mpc_file)
    net = from_ppc(ppc, f_hz, validate_conversion)

    return net

def _mpc2ppc(mpc_file, casename_mpc_file):
    # load mpc from file
    mpc = scipy.io.loadmat(mpc_file, squeeze_me=True, struct_as_record=False)

    # init empty ppc
    ppc = dict()

    _copy_data_from_mpc_to_ppc(ppc, mpc, casename_mpc_file)
    _adjust_ppc_indices(ppc)
    _change_ppc_TAP_value(ppc)

    return ppc

The function _mpc2ppc() is not same as the function on the github repo. Can someone help?

jwiemer112, 2023-01-23:
When you install pandapower via `pip install pandapower` the latest master release will be installed. I guess your cloned repo from this github page is checkout out on the `develop` branch. This could be the reason. 

You can also install different branches with pip, look at this:

https://stackoverflow.com/questions/20101834/pip-install-from-git-repo-branch

Does this solve your problem?

Greetz
Jan

jwiemer112, 2023-01-25:
@KishanGitASU Closing this issue, feel free to re-open it, if there is any need for it.

closed 2023-01-25 by jwiemer112

KishanGitASU, 2023-01-29:
`pip install pandapower==2.11.1` gives the function where the if loop for identifying the .mat, .m files. However, `pip install pandapower==2.2.2` gives the function that assumes the input file is always .mat file. I verified this on my end.