e2nIEE/pandapower #1835: plotly tutorials take 80% of the pandapower package size

SteffenMeinecke, 2023-01-23:
Three tutorial files take much storage space. Is there a possibility for reduction? @jkisse @rbolgaryn 
[plotly_traces](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/plotly_traces.ipynb) 41.7MB
[plotly_built-in](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/plotly_built-in.ipynb) 26.7MB
[plotly_maps](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/plotly_maps.ipynb) 15.8MB

### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

## Labels (if access available)
* [x] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

jkisse, 2023-01-23:
These tutorials are so large because of the plotly-HTML-outputs. But apparently, the output is not even displayed properly in the tutorial.
I'd propose we remove all outputs. Optionally, the output HTMLs could be uploaded in the pandapower-pages repo (the website) and hyperlinked in the tutorials. We could also include screenshots of the outputs. What do you think @rbolgaryn and @jwiemer112 ?

By the way, all ipynb-files are excluded in the pip-package

jwiemer112, 2023-01-23:
We could also replace the big `.html` plots with smaller png screenshots or make the renderer of the plots selectable ("svg", "browser", "png" and so on).

rbolgaryn, 2023-01-23:
is the problem that it is package size or is it that it takes up so much size when you check it out from develop?

rbolgaryn, 2023-07-19:
I think if the files are excluded in the pip package it is fine the way it is. 

If the outputs are incorrect, the tutorials should be adjusted, of course.

closed 2023-07-19 by rbolgaryn