e2nIEE/pandapower #1851: [bug] `net.res_bus['p_mw']` and `net.res_bus['q_mvar']` do not match `net._ppc['internal']['Sbus']`

christian-cahig, 2023-01-29:
### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [x] Tried basic troubleshooting (if a bug/error)
    * [x] Restarted interpreter
    * [x] Checked the pythonpath
* [x] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description

I was doing power flow with the 1888RTE power system test case as follows.

```python
import pandapower as pp
import pandapower.networks as pn
import numpy as np

net = pn.case1888rte(ref_bus_idx=1246) # Default reference bus
pp.runpp(net, calculate_voltage_angles=True)
```

Out of curiosity, I checked how `net.res_bus['p_mw']` and `net.res_bus['q_mvar']` (in per-unit values) relate to `net._ppc['internal']['Sbus']`,

```python
Sbus_ppc = net._ppc['internal']['Sbus']
Pbus_res = net.res_bus['p_mw'] / net._ppc['internal']['baseMVA']
Qbus_res = net.res_bus['q_mvar'] / net._ppc['internal']['baseMVA']
```

expecting that they should be (at least approximately) equal. It turns out they are not:

```python
>> np.allclose(-Sbus_ppc.real, Pbus_res), np.allclose(-Sbus_ppc.imag, Qbus_res)
(False, False)
>> np.linalg.norm(-Sbus_ppc.real-Pbus_res, ord=np.Inf), np.linalg.norm(-Sbus_ppc.imag-Qbus_res, ord=np.Inf)
(9.90900993480918, 2.735374118364189)
```

I also tried comparing `net.res_bus` values to the net injections calculated via the voltage phasors and the admittance matrix, i.e.,

```python
Sbus_eqn = net._ppc['internal']['V'] * ((net._ppc['internal']['Ybus'] * net._ppc['internal']['V']).conj())
```

and got lucky with the active component:

```python
>> np.allclose(-Sbus_eqn.real, Pbus_res), np.allclose(-Sbus_eqn.imag, Qbus_res)
(True, False)
>>  np.linalg.norm(-Sbus_eqn.real-Pbus_res, ord=np.Inf), np.linalg.norm(-Sbus_eqn.imag-Qbus_res, ord=np.Inf)
(8.139267038131948e-12, 2.322358397032247)
```

For completeness, I also compared the net injections via `net._ppc['internal']['Sbus']` and those calculated:

```python
>> np.allclose(Sbus_ppc.real, Sbus_eqn.real), np.allclose(Sbus_ppc.imag, Sbus_eqn.imag)
(False, False)
>> np.linalg.norm(Sbus_eqn.real-Sbus_ppc.real, ord=np.Inf), np.linalg.norm(Sbus_eqn.imag-Sbus_ppc.imag, ord=np.Inf)
(9.90900993480918, 2.735374118364189)
```

Help is much appreciated.

## Versions

* numpy version: 1.23.5
* pandapower version: 2.11.1
* pandas version: 1.5.3
* scipy version: 1.10.1
* Operating System name/version: Windows 10

rbolgaryn, 2023-02-03:
Dear @christian-cahig ,

this is an interesting observation, thank you.

AnkurArohi, 2023-07-05:
@christian-cahig Are you sure that the comparision being made are for the same bus indices/ same bus?
It could be that the S , P, Q are maybe not for the same bus, I think that the indexing sequence for the bus could be important in this case. I mean to say , _ppc and res_bus indexing could be different/ have a different starting value 0 or 1 (offset)

vogt31337, 2026-11-13:
### 1. The Q gap between res_bus and V·conj(Ybus·V) comes from the shunts.

- The 1888rte case has 45 shunts. In the ppc they are stored in the bus table as admittances (GS/BS), so they become part of Ybus.
- That means V·conj(Ybus·V) is the injection without the shunts. res_bus.q_mvar is the sum of all elements at the bus, shunts included.
- After adding the shunt power -BS·|V|² back, the Q error drops from 2.32 to 1.3e-11.
- P already matched because these shunts have no conductance (total shunt P is 0 MW).

### 2. The gap between Sbus and the actual injection is expected.

- ppci["internal"]["Sbus"] holds the injections specified as solver input, not the solved results.
- Every mismatch is at a bus where the power flow solves for that value: Q at the 271 PV buses, and P and Q at the one slack bus. All PQ buses match exactly.
- The solved slack P and generator Q are written to the result tables, not back into Sbus.

closed 2026-11-13 by vogt31337