e2nIEE/pandapower #1854: Sgen Cost has an effect on opf converging or not

pajjaecat, 2023-01-30:
Hello everyone!

I've been having an issue that I would like your help with. 

I have a custom PandaPower Network that is working Fine. In this network, I have a bunch of controllable Sgen. 
The opf objective is to Maximize generation (Minimize curtailment) while respecting the constraints ``max_vm_pu`` and ``min_vm_pu`` on several buses.  To achieve that,  I've created a negative poly_cost ``cp1_eur_per_mw=-1`` for all controllable Sgen in the network. 

Everything works fine using pf but the opf does not converge. I've set ``calculates_voltage_angles = False``  as indicated in #489 but it did not help. 
On the road of banging my head to find a solution, I've started messing with the poly_cost.  What has ended up working is that I've keep one of the sgen cost at -1 and reduced the other ones ( see bellow
[image]
). 

My question is,  how does a different choice of cost impact opf convergence? Is there something I'm missing about the costs? 

I've checked the issues here; however, nothing seems to be related to my questions. 

Thanks in advance for your answers

Ps: It is my first time posting here, so just bear with me.

#OPF

jwiemer112, 2023-02-01:
Please use the new issue template

pajjaecat, 2023-02-03:
Hey @jwiemer112 . 

Sorry I did not know about the new template.

Is there something I need to add or remove ??

jwiemer112, 2023-02-03:
if you open a new issue, choose a template. You should see the following:

[image]

Please edit this issue based on a templare, or close it and reopen an issue with a template.

pajjaecat, 2023-02-03:
Thanks, @jwiemer112  I've just updated the issue following the template. 

I've done some work since I've opened the issue. 

On the opf parameters, I've decreased  `delta`, set ``calculate_voltage_angles=False`` and changed ``init`` to the different possible values, but nothing worked. 

I've read on the Pandapower website, "The optimization with PYPOWER functionality does not have the best convergence properties", so I've tried with PandaModels.jl, and the problem has converged. However, the solutions are far from optimal. The constraints are respected, but the objective, which is to maximize the controllable Sgen generation is not respected.  

Please, Any tips would be highly appreciated.

jwiemer112, 2023-02-03:
Can you post a minimal code example?

rbolgaryn, 2023-03-15:
Hi @ZhengLiu1119 ,

do you have any comments about PandaModels?

ZhengLiu1119, 2023-03-23:
> 

sorry for the late response. After giving the code example I will test it and try to answer his question.

ZhengLiu1119, 2023-03-23:
> Hi @ZhengLiu1119 ,
> 
> do you have any comments about PandaModels?

sorry for the late response, I will test it with pandamodels using his code example (not given yet) and try to find the answer to his question.

pajjaecat, 2023-03-23:
Hello all. I'm really sorry for answering late. 

I'm currently really busy with other projects not related to this question. I will post a working code ASAP. 

Thanks in advance!

pajjaecat, 2023-04-25:
Hi @jwiemer112, @ZhengLiu1119 . 

I'm finally back here. 

So in order to find a solution to the previously described problem, I've run an opf on the exact same network using both pypower and powerModel. However, they do not yield the same results.

Here is the network used in the code below [testNetExcel.xlsx](https://github.com/e2nIEE/pandapower/files/11324446/testNetExcel.xlsx)

Here is my code.

```python
import pandapower as pp
import copy
import matplotlib.pyplot as plt

# Load the network 
# The network is already initialized such that to maximize injection of controllable sgens
# while respecting maximum and minimum voltage on certain buses. 
testNet = pp.from_excel('testNetExcel.xlsx')

# Create two copy of the network, one to be used with pypower and the second with powerModels.jl
net_pypower = copy.deepcopy(testNet)
net_julia = copy.deepcopy(testNet)

# run opf using both pypower and powerModel.jl
pp.runopp(net_pypower)
pp.runpm_ac_opf(net_julia)

# Plot maximum power on the controllable Sgen before and after opf using both methods
# Figure 1 
net_pypower.sgen.max_p_mw.plot(lw=0.7,marker="*", label='Max Power Available')
net_pypower.res_sgen.p_mw.plot(lw=0.7, marker="*", label='Power with PyPower')
net_julia.res_sgen.p_mw.plot(lw=0.7, marker="*", label='Power with PowerModel.jl')
plt.xlabel('Controllable Sgen number')
plt.legend()

markerSize = 5
max_vm_hv = 1.075  # Maximum voltage on Hv buses
max_vm_lv = 1.025   # Maximum voltage on Lv buses

# PLot maximum voltage on buses to check if constraints are respected
# Figure 2
plt.axhline(max_vm_hv, color='red', alpha=1, lw=0.5)
net_pypower.res_bus.loc[net_pypower.bus.query('max_vm_pu==@max_vm_hv').index].vm_pu.plot(marker='*', 
                                                                            ls='',
                                                                            markersize=markerSize, 
                                                                            ylabel='vm_pu', 
                                                                            xlabel='Higher Voltage Buses number', 
                                                                            label= 'PyPower', color='tab:orange')
net_julia.res_bus.loc[net_julia.bus.query('max_vm_pu==@max_vm_hv').index].vm_pu.plot(marker='*',
                                                                            ls='',
                                                                            markersize=markerSize, 
                                                                            label= 'PowerModels.jl', 
                                                                            color='tab:green')
plt.legend()

# Figure 3
plt.axhline(max_vm_lv, color='red', alpha=1, lw=0.5, markersize=0.5)
net_pypower.res_bus.loc[net_pypower.bus.query('max_vm_pu==@max_vm_lv').index].vm_pu.plot(marker='*', 
                                                                                ls='',
                                                                                markersize=markerSize, 
                                                                                ylabel='vm_pu', 
                                                                                xlabel='Lower Voltage Buses number', 
                                                                                label= 'PyPower', color='tab:orange')
net_julia.res_bus.loc[net_julia.bus.query('max_vm_pu==@max_vm_lv').index].vm_pu.plot(marker='*', 
                                                                                     ls='', 
                                                                                     label= 'PowerModels.jl', 
                                                                                     color='tab:green')
plt.legend()

```

### Figure 1
[image]

### Figure 2
[image]

### Figure 3
[image]

Any tips to make the PowerModels results as good as the pypower's would be highly appreciated. 

Thanks in advance !

jwiemer112, 2023-06-14:
Hey @ZhengLiu1119, @pajjaecat is this issue still relevant or can it be closed?

pajjaecat, 2023-06-14:
Hi @jwiemer112. It is still relevant ! I didn't find a solution yet !

rbolgaryn, 2023-07-19:
Hi @pajjaecat,

is the issue the differences between the results of pypower OPF and PandaModels OPF?

pajjaecat, 2023-07-19:
Hi @rbolgaryn . 

Yes the issue is why the difference between both methods ? 

Especially how can I make the PowerModels results as good as the pypower's , given that the former seems to have  better convergence property compared to the latter.

ZhengLiu1119, 2023-07-21:
hello @pajjaecat,

thank you for your code. I have just tested it locally. The results for pypower and PandaModels are almost the same, see below. 

[image]
[image]
[image]

which version of PandaModels are you using? the corrent version is 0.7.2

pajjaecat, 2023-08-17:
Hi @ZhengLiu1119  . 

Thank you for your feedback. 

I can't tell you which version of pandaModels I was using. I lost the hard drive
on my laptop, so I was compelled to reinstall everything from scratch. After re-installing,
I've retested my code with  PandaModels v0.7.2. and got the same results as you did. 

Thanks again.

closed 2023-08-17 by pajjaecat