e2nIEE/pandapower #1860: Problem when simulating an asymetric powerflow

DiegoCanas, 2023-02-03:
Hi! I was trying to simulate  an asymetric powerflow when I got this error: _**_Calculation of 3-phase power flow is only implemented for the transformer vector groups 'YNyn', 'Dyn', 'Yzn'_**_

I think the function of the transformer I made is ok, (the values are not real, I was going to do some tests before)

This is my full code:

#Import Pandapower
import pandapower as pp 
import pandapower.networks
from pandapower.pf.runpp_3ph import runpp_3ph

#Empty network creation
net = pp.create_empty_network() 

#Bus creation
bus1 = pp.create_bus(net, name="HV Busbar", vn_kv=110, type="b")
bus2 = pp.create_bus(net, name="HV Busbar 2", vn_kv=110, type="b")
bus3 = pp.create_bus(net, name="HV Transformer Bus", vn_kv=110, type="n")
bus4 = pp.create_bus(net, name="MV Transformer Bus", vn_kv=20, type="n")
bus5 = pp.create_bus(net, name="MV Main Bus", vn_kv=20, type="b")
bus6 = pp.create_bus(net, name="MV Bus 1", vn_kv=20, type="b")
bus7 = pp.create_bus(net, name="MV Bus 2", vn_kv=20, type="b")

#Bus table
busT = net.bus
print(busT)

#External Grid creation
pp.create_ext_grid(net, bus1, vm_pu=1.02, va_degree=50, name = "External Grid")
net.ext_grid['s_sc_max_mva'] = 1000
net.ext_grid['rx_max'] = 0.1
net.ext_grid['x0x_max']=0.1
net.ext_grid['r0x0_max']=0.1
pp.add_zero_impedance_parameters(net)

#External Grid table
exGridT = net.ext_grid
print(exGridT)

#Transformer Creation
trafo1 = pp.create_transformer_from_parameters(net, bus3, bus4,sn_mva=10, vn_hv_kv=110, vn_lv_kv=20, vkr_percent=1, vk_percent=1,pfe_kw=0.01,i0_percent=0.1, vector_group='Yzn',vk0_percent=1, vkr0_percent=1, mag0_percent=1, mag0_rx=1, si0_hv_partial=1, name="110kV/20kV transformer", std_type="25 MVA 110/20 kV")

#Transformer table
TrafoT = net.trafo
print (TrafoT)

#Line Creation

line1 = pp.create_line_from_parameters(net, bus1, bus2, length_km=10, r_ohm_per_km=0.00001, x_ohm_per_km=0.00001,c_nf_per_km=0.00002,r0_ohm_per_km=0.000001, x0_ohm_per_km=0.000001, c0_nf_per_km=0.000002, max_i_ka=0.8, std_type="N2XS(FL)2Y 1x300 RM/35 64/110 kV",  name="Line 1")
line2 = pp.create_line_from_parameters(net, bus5, bus6, length_km=2.0, r_ohm_per_km=0.00001, x_ohm_per_km=0.00001,c_nf_per_km=0.00002,r0_ohm_per_km=0.000001, x0_ohm_per_km=0.000001, c0_nf_per_km=0.000002, max_i_ka=0.8, std_type="NA2XS2Y 1x240 RM/25 12/20 kV", name="Line 2")
line3 = pp.create_line_from_parameters(net, bus6, bus7, length_km=3.5, r_ohm_per_km=0.00001, x_ohm_per_km=0.00001,c_nf_per_km=0.00002,r0_ohm_per_km=0.000001, x0_ohm_per_km=0.000001, c0_nf_per_km=0.000002, max_i_ka=0.8, std_type="48-AL1/8-ST1A 20.0", name="Line 3")
line4 = pp.create_line_from_parameters(net, bus7, bus5, length_km=2.5, r_ohm_per_km=0.00001, x_ohm_per_km=0.00001,c_nf_per_km=0.00002,r0_ohm_per_km=0.000001, x0_ohm_per_km=0.000001, c0_nf_per_km=0.000002, max_i_ka=0.8, std_type="NA2XS2Y 1x240 RM/25 12/20 kV", name="Line 4")

#Line Table
LineT = net.line
print (LineT)

#Switches Creation
#Switches Bus-Bus
sw1 = pp.create_switch(net, bus2, bus3, et="b", type="CB", closed=True)
sw2 = pp.create_switch(net, bus4, bus5, et="b", type="CB", closed=True)
#Switchers Bus-Line
sw3 = pp.create_switch(net, bus5, line2, et="l", type="LBS", closed=True)
sw4 = pp.create_switch(net, bus6, line2, et="l", type="LBS", closed=True)
sw5 = pp.create_switch(net, bus6, line3, et="l", type="LBS", closed=True)
sw6 = pp.create_switch(net, bus7, line3, et="l", type="LBS", closed=False)
sw7 = pp.create_switch(net, bus7, line4, et="l", type="LBS", closed=True)
sw8 = pp.create_switch(net, bus5, line4, et="l", type="LBS", closed=True)

#Switch Table
SwitchT = net.switch
print(SwitchT)

#Load Creation
pp.create_asymmetric_load(net, bus7, p_a_mw=2, q_a_mvar=4, scaling=0.6, name="load", type="wye")
#Voltage dependant load, we define what percentage of load  is depndeant on constant power, current or impedance
pp.create_load(net, bus7, p_mw=2, q_mvar=4, const_z_percent=30, const_i_percent=20, name="zip_load")

#Load Table
LoadT = net.load
print(LoadT)

#Static Generator Creation
pp.create_sgen(net, bus7, p_mw=2, q_mvar=-0.5, name="static generator")

#Static Generator Table
SGenT = net.sgen
print (SGenT)

#Voltage Controlled Generator Creation
pp.create_asymmetric_sgen(net, bus6, p_a_mw=6, q_a_mvar=3, name="generator")

#Voltage Controlled Generator Table
VGenT = net.gen
print (VGenT)

#Shunt Creation
pp.create_shunt(net, bus3, q_mvar=-0.96, p_mw=0, name='Shunt')

#Shunt Table
ShuntT = net.shunt
print (ShuntT)

#POWER FLOW ANALYSIS

#Power flow Unbalanced run
pp.add_zero_impedance_parameters(net)
runpp_3ph(net)
#pandapower.pf.runpp_3ph.runpp_3ph(net, calculate_voltage_angles=True, init='auto', max_iteration='auto', tolerance_mva=1e-08, trafo_model='t', trafo_loading='current', enforce_q_lims=False, numba=True, recycle=None, check_connectivity=True, switch_rx_ratio=2.0, delta_q=0, v_debug=False)

Thank you for your help!

rbolgaryn, 2023-02-03:
The function pp.add_zero_impedance_parameters(net) calls this function: parameter_from_std_type(net, "vector_group", element="trafo"), which overwrites your specified vector_group with a different one from the std_type library.

Try creating a new transformer type and use "create_transformer", or don't set the std_type name when you create a transformer.

Best regards,
Roman

DiegoCanas, 2023-02-06:
Thank you very much, now it works perfect!

rbolgaryn, 2023-03-15:
you are welcome!

closed 2023-03-15 by rbolgaryn