e2nIEE/pandapower #1862: Plotting: Creating Single-Line

LukasKur, 2023-02-04:
Hey guys,

I´m just starting with pandapower, but I don´t find a function for that in the Docs:
Is it possible to plot single-line-diagrams with pandapower (e.g. like in DigSilent Powerfactory)?
I only need the structure of the network, not the information about the geodata.

Something like that would be great:
[image]

Thanks in advance and kinds regards,
Lukas

jwiemer112, 2023-02-07:
Hey Lukas, please use the [Issue-Templates](https://github.com/e2nIEE/pandapower/issues/new/choose).

At the moment pandapower offers plotting functions based on **matplotlib** and **plotly**. These visualize the network topology based on geocoordinates.

An exact electrical map of the network cannot be plotted easily at the moment. We are open for suggestions, collaborations and pull requests from the community.

Greetz Jan

nicorikken, 2023-02-07:
Best free software implementation I've seen so far is in PowSyBl https://github.com/powsybl/powsybl-single-line-diagram-server Could be inspiration for an implementation in Python, or perhaps those developers are willing to collaborate.

LukasKur, 2023-02-11:
Alright, thanks a lot!

closed 2023-02-11 by LukasKur