e2nIEE/pandapower #1864: [feature] Plotly | Scale Legend | Weighted marker trace

jwiemer112, 2023-02-06:
### Feature Checklist
* [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests
* [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

**Describe the solution you'd like**
Calculate the **scale legend** position dynamically instead of static. Show the related **sizemode** in the legend.
Minimize overlapping.

**Additional context**
[image]

## Labels (if access available)
* [X] Choose a `labels` at the right side (e.g. `controller`, `OPF`...)

rbolgaryn, 2023-07-19:
Hi @jwiemer112 ,

what is the status of the issue?

vogt31337, 2024-11-21:
I think pie slices signifying the situation on each bus would be more interesting. @jwiemer112

closed 2025-11-27 by KS-HTK