e2nIEE/pandapower #1866: ShortCircuit caluclation doesn´t work

LukasKur, 2023-02-11:
Hey,

I´m just experimenting with pandapower and now trying to run some short-circuit calculations with the following code:
(I changed some names cause of data security, in the real code all variables are named correctly and e.g. power flow works totally fine)

```
`
import pandapower as pp
import pandapower.plotting as plot
from numpy import pi
import os
import matplotlib.pyplot as plt
from pandapower.plotting.plotly import simple_plotly

%matplotlib inline

#Netzwerk einfügen
net = pp.create_empty_network()

#Sammelschienen hinzufügen
HVW = pp.create_bus(net, 110, "HVW1") 
GuD = pp.create_bus(net, 110, "SS_GuD") 

BAC41 = pp.create_bus(net, 11, "41") 
BAC42 = pp.create_bus(net, 11, "42") 

net.bus

#Stromnetz hinzufügen
pp.create_ext_grid(net, HVW, s_sc_max_mva = 3926.7, s_sc_min_mva = 3800, rx_max = 0.1, rx_min = 0.05)

#Kabel hinzufügen
pp.create_line(net,from_bus = 0, to_bus = 1, length_km=0.1,std_type="N2XS(FL)2Y 1x300 RM/35 64/110 kV")

#Transformatoren hinzufügen
pp.create_transformer3w_from_parameters(net, GuD, 41, 42, name="40", 
                                        vn_hv_kv = 110, 
                                        vn_mv_kv = 10, 
                                        vn_lv_kv = 10,
                                        sn_hv_mva = 136,
                                        sn_mv_mva = 68,
                                        sn_lv_mva = 68,
                                        vk_hv_percent = 11,
                                        vk_mv_percent = 22, 
                                        vk_lv_percent = 11,
                                        vkr_hv_percent = 8,
                                        vkr_mv_percent = 19,
                                        vkr_lv_percent = 8,
                                        pfe_kw =300,
                                        i0_percent = 2.9e-4)
net.trafo3w

#Generatoren hinzufügen
pp.create_gen(net, 41,
                 p_mw = -50.2,
                 sn_mva = -62.75,
                 vn_kv = 11.5,
                 xdss_pu = 0.178,
                 rdss_ohm = 0.177,
                 cos_phi = 0.8,
                 pg_percent = 1.1,
                 power_station_trafo = 0)

pp.create_gen(net, 42,
                 p_mw = -50.2,
                 sn_mva = -62.75,
                 vn_kv = 11.5,
                 xdss_pu = 0.178,
                 rdss_ohm = 0.177,
                 cos_phi = 0.8,
                 pg_percent = 1.1,
                 power_station_trafo = 0)

#Lasten hinzufügen
pp.create_load(net,41,p_mw=10, q_mvar = 1, name="WP")

#Kabel hinzufügen
pp.create_line(net,HVW,GuD,2,"N2XS(FL)2Y 1x300 RM/35 64/110 kV") 

pp.shortcircuit.calc_sc(net, fault="3ph", case="max")
`
```

But when running the SC-calc I get this error:
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[34], line 1
----> 1 pp.shortcircuit.calc_sc(net, fault="3ph", case="max")

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:147, in calc_sc(net, bus, fault, case, lv_tol_percent, topology, ip, ith, tk_s, kappa_method, r_fault_ohm, x_fault_ohm, branch_results, check_connectivity, return_all_currents, inverse_y)
    144 init_results(net, "sc")
    146 if fault in ("2ph", "3ph"):
--> 147     _calc_sc(net, bus)
    148 elif fault == "1ph":
    149     _calc_sc_1ph(net, bus)

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:196, in _calc_sc(net, bus)
    195 def _calc_sc(net, bus):
--> 196     ppc, ppci = _init_ppc(net)
    198     _calc_current(net, ppci, bus)
    200     ppc = _copy_results_ppci_to_ppc(ppci, ppc, "sc")

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\ppc_conversion.py:42, in _init_ppc(net)
     40 # Add parameter K into ppc
     41 _add_kt(net, ppc)
---> 42 _add_gen_sc_z_kg_ks(net, ppc)
     43 _add_xward_sc_z(net, ppc)
     45 ppci = _ppc2ppci(ppc, net)

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\ppc_conversion.py:154, in _add_gen_sc_z_kg_ks(net, ppc)
    152 # Calculate K_S on power station configuration
    153 if np.any(~np.isnan(gen.power_station_trafo.values)):
--> 154     f, _ = net["_pd2ppc_lookups"]["branch"]["trafo"]
    156     # If power station units defined with index in gen, no topological search needed
    157     ps_gen_mask = ~np.isnan(gen.power_station_trafo.values)

KeyError: 'trafo'
```

What did I wrong?

Thanks in advance and kind regards,
Lukas

jwiemer112, 2023-02-14:
Hey Lukas,

as mentioned in your other Issue, please use our Issue-Templates. The information is required in order to be able to manage issues more clearly and to be able to process them better.

Greetz
Jan

LukasKur, 2023-02-14:
Hey Jan,

I thought I made it right this time, it wasn't on purpose. Sorry!
But in the templates I only saw one for Bugs and one for features, and I wouldn't sort it into one of them.

Regards, Lukas

LukasKur, 2023-02-17:
Ah, I understand. Sorry, I´m quite new to this topic.
I will close it and reopen it new.

closed 2023-02-17 by LukasKur