e2nIEE/pandapower #1875: [feature] simple_plot | Make Orientation of Assets Adjustable

jwiemer112, 2023-02-14:
At the moment the orientation of assets in the pandapower plotting module `simple_plot` is hard coded to 0. 
We should enhance the plotting experience and make it adjustable. We could also discuss to change some of the symbols.

### Feature Checklist
* [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests
* [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

**Describe the solution you'd like**
Make the orientation of assets adjustable in `simple_plot`

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[image]
[image]

## Labels (if access available)
* [X] Choose a `labels` at the right side (e.g. `controller`, `OPF`...)

jwiemer112, 2023-03-17:
@hzraja

rbolgaryn, 2023-07-19:
Hello @hzraja ,

what is the status of this issue?

vogt31337, 2024-11-21:
corresponds to https://github.com/e2nIEE/pandapower/pull/2114 ?