e2nIEE/pandapower #1877: [bug] from_mpc() not working for case14 and produce warning for many cases

amir-saadat, 2023-02-17:
Hi,

I don't get reliable results from `from_mpc`. For example, the following produces an error

```
import pandapower.converter as ppc
net = ppc.from_mpc('case14.m', f_hz=60)
```

error:
```
/Users/amirsaadat/.pyenv/versions/3.10.10/lib/python3.10/site-packages/pandapower/converter/pypower/from_ppc.py:209: RuntimeWarning: invalid value encountered in divide
  max_i_ka = ppc['branch'][:, 5]/ppc['bus'][to_bus, BASE_KV]/np.sqrt(3)
/Users/amirsaadat/.pyenv/versions/3.10.10/lib/python3.10/site-packages/pandapower/converter/pypower/from_ppc.py:220: RuntimeWarning: divide by zero encountered in divide
  c_nf_per_km=(ppc['branch'][is_line, BR_B]/Zni[is_line]/omega*1e9/2).real,
/Users/amirsaadat/.pyenv/versions/3.10.10/lib/python3.10/site-packages/pandapower/converter/pypower/from_ppc.py:220: RuntimeWarning: invalid value encountered in divide
  c_nf_per_km=(ppc['branch'][is_line, BR_B]/Zni[is_line]/omega*1e9/2).real,
There are 3 branches which are considered as trafos - due to ratio unequal 0 or 1 - but connect same voltage levels.
```

In addition, cases greater than and equal to 14 produce the last warning message that I am not entirely sure if it is concerning. 

I am using the latest version of the pandapower.

```
matpower           7.1.0.2.1.4
matpowercaseframes 1.0.4
numba              0.56.4
numpy              1.23.5
pandapower         2.11.1
```

SteffenMeinecke, 2023-05-09:
Hi @amir-saadat ,
the error message is logging what the issue is. Do you doubt the correctness of the message or what is your point?

rbolgaryn, 2023-07-19:
closing due to inactivity.

closed 2023-07-19 by rbolgaryn