e2nIEE/pandapower #1902: [bug] SC-calculation doesn´t work with 2 or more transformators

LukasKur, 2023-03-12:
### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [x] Tried basic troubleshooting (if a bug/error)
    * [x] Restarted interpreter
    * [x] Checked the pythonpath
* [x] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description
Hey all,
I´m trying to make a short-circuit-calculation. I made a previous issue, where I described the error in my main project #1876 .
Now I was trying out some examples/tutorials to test the problem.
I tried [https://github.com/e2nIEE/pandapower/blob/develop/tutorials/shortcircuit.ipynb](url) and it worked.
But when I add a second transformator (see code) below I get the error below.

--> Is the problem/bug of the short-circuit-calculation the second transformator? 
(FYI, the example in [https://pandapower.readthedocs.io/en/v2.11.1/shortcircuit/run.html](url) doesn´t work with me.

### What steps reproduce the problem?

Executing the following code in Jupyter Notebook:
`import pandapower as pp
import pandapower.shortcircuit as sc
import pandapower.plotting as plot

from numpy import pi

try:
    import seaborn
    colors = seaborn.color_palette()
except:
    colors = ["b", "g", "r", "c", "y"]

def ring_network():
    net = pp.create_empty_network()
    b1 = pp.create_bus(net, 220)
    b2 = pp.create_bus(net, 110)
    b3 = pp.create_bus(net, 110)
    b4 = pp.create_bus(net, 110)
    
    pp.create_ext_grid(net, b1, s_sc_max_mva=100., s_sc_min_mva=80., rx_min=0.20, rx_max=0.35)
    
    pp.create_transformer(net, b1, b2, "100 MVA 220/110 kV")
    pp.create_transformer(net, b2, b3, "100 MVA 220/110 kV")
    
    #pp.create_line(net, b2, b3, std_type="N2XS(FL)2Y 1x120 RM/35 64/110 kV" , length_km=15.)
    l2 = pp.create_line(net, b3, b4, std_type="N2XS(FL)2Y 1x120 RM/35 64/110 kV" , length_km=12.)
    pp.create_line(net, b4, b2, std_type="N2XS(FL)2Y 1x120 RM/35 64/110 kV" , length_km=10.)
    pp.create_switch(net, b4, l2, closed=False, et="l")
    return net

net = ring_network()

sc.calc_sc(net, case="max", ip=True, ith=True, branch_results=True)
net.res_bus_sc`

### What is the expected output? What do you see instead?

Short-circuit calculation

### Paste Traceback/Error Below (if applicable)

```python-traceback

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[3], line 1
----> 1 sc.calc_sc(net, case="max", ip=True, ith=True, branch_results=True)
      2 net.res_bus_sc

AttributeError: 'PatchCollection' object has no attribute 'calc_sc'

```

## Versions

pandas version: 1.5.2
networkx version: 3.0
scipy version: 1.10.0
numpy version: 1.23.5
packaging version: 22.0
tqdm version: 4.64.1
deepdiff version: not included --> reason for the error?
Operating System name/version: Win 10

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

LukasKur, 2023-03-13:
**Update**
I updated pandapower (with conda, if this information is helpful) again and now I get the following error
when executing the code above:

```
`---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[11], line 1
----> 1 pp.shortcircuit.calc_sc(net, fault="3ph", case="max")

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:147, in calc_sc(net, bus, fault, case, lv_tol_percent, topology, ip, ith, tk_s, kappa_method, r_fault_ohm, x_fault_ohm, branch_results, check_connectivity, return_all_currents, inverse_y)
    144 init_results(net, "sc")
    146 if fault in ("2ph", "3ph"):
--> 147     _calc_sc(net, bus)
    148 elif fault == "1ph":
    149     _calc_sc_1ph(net, bus)

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:198, in _calc_sc(net, bus)
    195 def _calc_sc(net, bus):
    196     ppc, ppci = _init_ppc(net)
--> 198     _calc_current(net, ppci, bus)
    200     ppc = _copy_results_ppci_to_ppc(ppci, ppc, "sc")
    201     _extract_results(net, ppc, ppc_0=None, bus=bus)

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:160, in _calc_current(net, ppci_orig, bus)
    156 ppci_bus = _get_is_ppci_bus(net, bus)
    158 # update ppci
    159 non_ps_gen_ppci_bus, non_ps_gen_ppci, ps_gen_bus_ppci_dict =\
--> 160     _create_k_updated_ppci(net, ppci_orig, ppci_bus=ppci_bus)
    162 # For each ps_gen_bus one unique ppci is required
    163 ps_gen_ppci_bus = list(ps_gen_bus_ppci_dict.keys())

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\ppc_conversion.py:245, in _create_k_updated_ppci(net, ppci_orig, ppci_bus)
    243         trafo_ix = ppci_gen["bus"][bus, PS_TRAFO_IX].astype(int)
    244         # Calculating SC inside power system unit
--> 245         ppci_gen["branch"][trafo_ix, [BR_X, BR_R]] /= \
    246             ppci_gen["branch"][trafo_ix, K_ST]
    248         bus_ppci.update({bus: ppci_gen})
    250 return non_ps_gen_bus, ppci, bus_ppci

IndexError: index -2147483648 is out of bounds for axis 0 with size 5`
````---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[11], line 1
----> 1 pp.shortcircuit.calc_sc(net, fault="3ph", case="max")

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:147, in calc_sc(net, bus, fault, case, lv_tol_percent, topology, ip, ith, tk_s, kappa_method, r_fault_ohm, x_fault_ohm, branch_results, check_connectivity, return_all_currents, inverse_y)
    144 init_results(net, "sc")
    146 if fault in ("2ph", "3ph"):
--> 147     _calc_sc(net, bus)
    148 elif fault == "1ph":
    149     _calc_sc_1ph(net, bus)

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:198, in _calc_sc(net, bus)
    195 def _calc_sc(net, bus):
    196     ppc, ppci = _init_ppc(net)
--> 198     _calc_current(net, ppci, bus)
    200     ppc = _copy_results_ppci_to_ppc(ppci, ppc, "sc")
    201     _extract_results(net, ppc, ppc_0=None, bus=bus)

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\calc_sc.py:160, in _calc_current(net, ppci_orig, bus)
    156 ppci_bus = _get_is_ppci_bus(net, bus)
    158 # update ppci
    159 non_ps_gen_ppci_bus, non_ps_gen_ppci, ps_gen_bus_ppci_dict =\
--> 160     _create_k_updated_ppci(net, ppci_orig, ppci_bus=ppci_bus)
    162 # For each ps_gen_bus one unique ppci is required
    163 ps_gen_ppci_bus = list(ps_gen_bus_ppci_dict.keys())

File ~\Anaconda3\envs\pypsaenv\lib\site-packages\pandapower\shortcircuit\ppc_conversion.py:245, in _create_k_updated_ppci(net, ppci_orig, ppci_bus)
    243         trafo_ix = ppci_gen["bus"][bus, PS_TRAFO_IX].astype(int)
    244         # Calculating SC inside power system unit
--> 245         ppci_gen["branch"][trafo_ix, [BR_X, BR_R]] /= \
    246             ppci_gen["branch"][trafo_ix, K_ST]
    248         bus_ppci.update({bus: ppci_gen})
    250 return non_ps_gen_bus, ppci, bus_ppci

IndexError: index -2147483648 is out of bounds for axis 0 with size 5`

rbolgaryn, 2023-03-14:
Dear @LukasKur ,

we do not have a conda version of pandapower, I think it is an older version packaged by someone else?

You should install pandapower from PyPI or from github develop. 
Does the problem still occur then?

LukasKur, 2023-03-14:
Hey,

thanks for the comment!
I will try it out.

FYI:
In the official  installation guide conda is explicitly recommended:
[http://www.pandapower.org/start/](http://www.pandapower.org/start/)

rbolgaryn, 2023-03-14:
Yes but for installing packages, not pandapower itself

rbolgaryn, 2023-07-04:
No updates so closing the issue.

closed 2023-07-04 by rbolgaryn