e2nIEE/pandapower #1909: [bug] Network converted from CIM Scheme showing multiple columns for bus_geodata. Not sure the data issue or grouping issue or both

AIShankhoGhosh, 2023-03-20:
### Issue Report Checklist

* [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for useful code snippets and examples of use

* [ ] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
CIM converter not yet in pip install

* [X] Tried basic troubleshooting (if a bug/error)
    * [X] Restarted interpreter
    * [X] Checked the pythonpath
* [ ] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description

### What steps reproduce the problem?

1. 
2. 
3. 

### What is the expected output? What do you see instead?

### Paste Traceback/Error Below (if applicable)

```python-traceback

PASTE TRACEBACK HERE

```

## Versions

* pandas version: 
* networkx version: 
* scipy version: 
* numpy version: 
* packaging version: 
* tqdm version: 
* deepdiff version: 
* Operating System name/version: 

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

AIShankhoGhosh, 2023-03-20:
[image]
[image]
[image]

mrifraunhofer, 2023-07-19:
Hey AIShankhoGhosh, 
this issue is not reproduceable in version 2.13.1. Converting the provided CGMES dataset leads to the following result:
This pandapower network includes the following parameter tables:
   - bus (**13546** element)
   - load (1017 element)
   - sgen (1237 element)
   - gen (12 element)
   - switch (16047 element)
   - shunt (21 element)
   - ext_grid (21 element)
   - line (2319 element)
   - trafo (73 element)
   - trafo3w (80 element)
   - impedance (422 element)
   - ward (875 element)
   - characteristic (446 element)
   - controller (5 element)
   - bus_geodata (**13546** element)
   - characteristic_temp (1917 element)
The number of busses and geodata for busses is equal.

Suggestion: Update pandapower to version 2.13.1

Regarding the columns: Even busses may have more than one x,y coordinate (ie. Busbars). For the lines: No geodata provided in the dataset.

Note: The provided CGMES dataset is not valid! No loadflow possible due to missing data.

closed 2023-08-10 by rbolgaryn