e2nIEE/pandapower #1911: SystemError while trying to solve power flows

kosmylo, 2023-03-26:
### Issue Report Checklist

* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

I am trying to solve the power flows as follows:
```
import pandapower as pp
import pandapower.networks as nw
import pandas as pd
import numpy as np

net = nw.case33bw()
pp.runpp(net)
print(net.res_bus)
```
However, I am getting the following error:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
TypeError: expected dtype object, got 'numpy.dtype[int64]'

The above exception was the direct cause of the following exception:

SystemError                               Traceback (most recent call last)
<ipython-input-3-b5247745d6ec> in <module>
----> 1 pp.runpp(net)
      2 print(net.res_bus)

~\anaconda3\lib\site-packages\pandapower\run.py in runpp(net, algorithm, calculate_voltage_angles, init, max_iteration, tolerance_mva, trafo_model, trafo_loading, enforce_q_lims, check_connectivity, voltage_depend_loads, consider_line_temperature, run_control, distributed_slack, tdpf, tdpf_delay_s, **kwargs)
    239         _check_bus_index_and_print_warning_if_high(net)
    240         _check_gen_index_and_print_warning_if_high(net)
--> 241         _powerflow(net, **kwargs)
    242 
    243 

~\anaconda3\lib\site-packages\pandapower\powerflow.py in _powerflow(net, **kwargs)
     82 
     83     # ----- run the powerflow -----
---> 84     result = _run_pf_algorithm(ppci, net["_options"], **kwargs)
     85     # read the results (=ppci with results) to net
     86     _ppci_to_net(result, net)

~\anaconda3\lib\site-packages\pandapower\powerflow.py in _run_pf_algorithm(ppci, options, **kwargs)
    162             result = _run_bfswpf(ppci, options, **kwargs)[0]
    163         elif algorithm in ['nr', 'iwamoto_nr']:
--> 164             result = _run_newton_raphson_pf(ppci, options)
    165         elif algorithm in ['fdbx', 'fdxb', 'gs']:  # algorithms existing within pypower
    166             result = _runpf_pypower(ppci, options, **kwargs)[0]

~\anaconda3\lib\site-packages\pandapower\pf\run_newton_raphson_pf.py in _run_newton_raphson_pf(ppci, options)
     56         ppci, success, iterations, bus, gen, branch = _run_ac_pf_with_qlims_enforced(ppci, options)
     57     else:
---> 58         ppci, success, iterations = _run_ac_pf_without_qlims_enforced(ppci, options)
     59         # update data matrices with solution store in ppci
     60         bus, gen, branch = ppci_to_pfsoln(ppci, options)

~\anaconda3\lib\site-packages\pandapower\pf\run_newton_raphson_pf.py in _run_ac_pf_without_qlims_enforced(ppci, options)
    149     baseMVA, bus, gen, branch, ref, pv, pq, _, _, V0, ref_gens = _get_pf_variables_from_ppci(ppci)
    150 
--> 151     ppci, Ybus, Yf, Yt = _get_Y_bus(ppci, options, makeYbus, baseMVA, bus, branch)
    152 
    153     # compute complex bus power injections [generation - load]

~\anaconda3\lib\site-packages\pandapower\pf\run_newton_raphson_pf.py in _get_Y_bus(ppci, options, makeYbus, baseMVA, bus, branch)
    104     else:
    105         # build admittance matrices
--> 106         Ybus, Yf, Yt = makeYbus(baseMVA, bus, branch)
    107 
    108     return ppci, Ybus, Yf, Yt

~\anaconda3\lib\site-packages\pandapower\pf\makeYbus_numba.py in makeYbus(baseMVA, bus, branch)
    165     Yt = coo_matrix((Yt_x, (i, col_Y)), (nl, nb)).tocsr()
    166     Yx, Yj, Yp, nnz = gen_Ybus(Yf_x, Yt_x, Ysh, col_Y, f, t, np.argsort(f), np.argsort(t), nb, nl,
--> 167                                np.arange(nl, dtype=np.int64))
    168     Ybus = csr_matrix((np.resize(Yx, nnz), np.resize(Yj, nnz), Yp), (nb, nb))
    169     for Y in (Ybus, Yf, Yt):

SystemError: CPUDispatcher(<function gen_Ybus at 0x000001C968169438>) returned a result with an error set
```
Any idea of what is going wrong here? There was a similar issue that mentioned that by updating the packages the problem is solved, but it did not happen in my case.

rbolgaryn, 2023-07-04:
you can repeat by creating a fresh Python environment and installing pandapower, numba, etc from scratch.

You can try following the instructions on the pandapower website to install a development version of pandapower.

rbolgaryn, 2023-07-19:
closing due to inactivity. @kosmylo feel free to reopen if still relevant & you have more information about the issue.

closed 2023-07-19 by rbolgaryn