e2nIEE/pandapower #1917: [feature] clean up import

SteffenMeinecke, 2023-03-29:
### Feature Checklist
* [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests
* [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

**1. How to import**
Recommended ways of importing basic pandapower functionaliy, e.g. `runpp()` should look like (as it is)
```
# option 1
import pandapower as pp
pp.runpp()  # usage

# option 2
from pandapower import runpp
runpp()  # usage
```

Recommended ways of importing futher pandapower functionaliy, e.g. `from_ppc()` should look like
```
# option 1 (default)
import pandapower as pp
pp.converter.from_ppc()  # usage

# option 2
from pandapower import converter as cv
cv.from_ppc()  # usage

# option 3
from pandapower.converter import from_ppc()
from_ppc()  # usage
```

Not recommended for usage outside the pandapower package itself (for functions not starting with `_`):
```
from pandapower.converter.pypower.from_ppc import from_ppc
from_ppc()  # usage
```
Reason: pandapower might move or rename the file containing from_ppc

*Remark:* `pp.converter` should be first available with https://github.com/e2nIEE/pandapower/pull/1916

*TODOs*:

- [ ] add recommendation as above somewhere to the documentation
- [ ] proof and adjust whether docstrings and documentation .rst file propose import as recommended

**2. Remove availability of toolbox functions and some others with `import pandapower`**

 Will destroy users' code that uses functions in question from `import pandapower` -> deprecation warnings as needed, [see](https://pandapower.readthedocs.io/en/develop/about/deprecating.html)

*TODOs*:
1st:

- [ ] agree on which functions shouldn't be available with `import pandapower` anymore (toolbox, some of auxiliary, ...)
- [ ] import placeholders that warns the wrong import way but still imports the correct function
- [ ] add a test that 

2nd:

- [ ] import placeholders should raise DeprecationErrors with the wrong import way (to be merged after the first release after 1st)

vogt31337, 2024-11-21:
We should also discuss to maybe introduce a more modular approach. At the moment, if you import pandapower nearly everything gets loaded directly (since many __init__.py's). I would like to see an approach, where you need to import pandapower.timeseries and only then the timeseries imports are loaded.

SteffenMeinecke, 2024-11-25:
> I would like to see an approach, where you need to import pandapower.timeseries and only then the timeseries imports are loaded.

I guess to increase the import time? If one import pandapower once at the beginning of a calculation run, it should be no big deal right now, or am I wrong?

KS-HTK, 2024-12-19:
I started work on this by rewriting all internal files to not use `import pandapower` in most cases.
Now when clearing out all pandapower imports the main `__init__.py` the load time is down to 1.3sec of which 1.2sec is pandas. That is about what I would expect for the `dump_to_geojson` function.

But a totally empty `__init__.py` is not really a good option for pandapower. So we should decide upon which function and modules should be imported in each modules `__init__.py` and which are only available via direct imports.

KS-HTK, 2025-01-22:
with PR #2512 I've updated most code to use Option 3. I've decided to use this option as deciding what will and won't be included in the `__init__.py` of each module will take time and will lead to issues when using option 2.

The PR includes some fixes for the doc strings, but I have not looked trough each tutorial and docs page to verify that I have found all ocurrences. Most should be covered by the PR though.

Modifying any `__init__.py` should not break any code within pandapower, dependant code will be affected.
> Will destroy users' code that uses functions in question from import pandapower -> we should be clear that we really want that -> if yes, deprecation warnings are needed, [see](https://pandapower.readthedocs.io/en/develop/about/deprecating.html)

Writing deprecations for imports seems impossible at the time. I tried looking into trowing Deprecation Warnings when e.g. `import pandapower as pp` is used but could not find a way to do this as a normal deprecation warning and not raising an error.

thatguy7, 2026-10-16:
One option might be to define an explicit pandapower.api module/package that imports most things and keep the __init__.py minimalistic.

Then, people who use pandapower as a library, will import just the parts they care about without expensive import times.
```python
import pandapower as pp
pp.timeseries  # not working

import pandapower.timeseries
pandapower.timeseries # working
pp.timeseries # working as well
```

And people who use pandapower interactively e.g. in a notebook or simply don't care about the import overhead, can do:
```python
import pandapower.api as pp
pp.timeseries # working
```

On the other hand with python 3.15 there is the option to define lazy imports - which would make this mostly superfluous.

The import time of pandapower can be annoying, when using it with multiprocessing so it's imported over and over again.