e2nIEE/pandapower #1985: [bug] Invalid syntax in python 3.7 and pandapower 2.12

BDonnot, 2023-04-13:
### Issue Report Checklist

* [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports
* [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html) 
* [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use
* [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)
* [X] Tried basic troubleshooting (if a bug/error)
    * [X] Restarted interpreter
    * [X] Checked the pythonpath
* [X] Completed the **Problem Description**, **What steps reproduce the problem** and **Versions** sections below

## Problem Description

When installed from scratch in python 3.7 (pandapower 2.12) i got the following error:

```python
from pandapower.toolbox.result_info import *
```

This raises the error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File ".local/lib/python3.7/site-packages/pandapower/__init__.py", line 14, in <module>
    from pandapower.toolbox import *  # to be removed -> in future via package namespace available
  File ".local/lib/python3.7/site-packages/pandapower/toolbox/__init__.py", line 2, in <module>
    from pandapower.toolbox.result_info import *
  File "<fstring>", line 1
    (element_type=)
                 ^
SyntaxError: invalid syntax
```

### What steps reproduce the problem?

1.  install python 3.7 (3.7.16 for me but I guess any version works)
2.  install pandapower: `python3.7 -m pip install --upgrade pandapower`
3.  run `python3.7 -c "from pandapower.toolbox.result_info import *"`

### What is the expected output? What do you see instead?

### Paste Traceback/Error Below (if applicable)

```python-traceback
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File ".local/lib/python3.7/site-packages/pandapower/__init__.py", line 14, in <module>
    from pandapower.toolbox import *  # to be removed -> in future via package namespace available
  File ".local/lib/python3.7/site-packages/pandapower/toolbox/__init__.py", line 2, in <module>
    from pandapower.toolbox.result_info import *
  File "<fstring>", line 1
    (element_type=)
                 ^
SyntaxError: invalid syntax
```

## Versions

* pandas version:
* networkx version: 
* scipy version: 
* numpy version: 
* packaging version: 
* tqdm version: 
* deepdiff version: 
* Operating System name/version: 

Above informations are irrelevant for the bug, this is a problem with the fstring not support the syntax `f"{var = }"` in python 3.7 (introduced apparently in python 3.8) , as shown in this super simple example:

```
 > python3.7 -c "a = 2; print(f'{a = }')"
  File "<fstring>", line 1
    (a = )
       ^
SyntaxError: invalid syntax
```
I can find them for you if you want
## Labels (if access available)
* [ ] Choose a `labels` at the right side (e.g. `bug`, `controller`, `OPF`...)

SteffenMeinecke, 2023-04-14:
python 3.7 support has been dropped with the latest version since its [life is going to end soon](https://devguide.python.org/versions/). Please update your python installation.

BDonnot, 2023-04-14:
Hello,

This error arises in a unit test script. I'll make sure not to install latest version of pandapower for older python versions. I totally understand you drop support for 3.7 this makes total sense. 

A better solution would be not to release new version of pandapower on pypi for python 3.7 I think.

Indeed, as of now trying to install it on python 3.7 is impossible due to above-mentioned error.

SteffenMeinecke, 2023-04-14:
Sorry for complications but why is it so that a new pp version appears for py37? It is clearly released for >py38...
[image]

BDonnot, 2023-04-14:
Hello,

I don't know the `why` but I assure you that, when you run python 3.7 (I installed a clean version locally on purpose) and try to install pandapower, it finds (and install) a non working pandapower version (2.12 in this case) and not 2.11 (which is the latest supported as far as I understand)

You can do the set in a virtual machine, in a docker container or in your computer.

Benjamin

BDonnot, 2023-04-17:
I create a venv for python3.7 from scratch, installed pandapower and you can see it installs pandapower 2.12 which should not happen:

[image]

SteffenMeinecke, 2023-04-19:
should be fixed now. please reopen if some problems occur.